chipsalliance/rocket-chip #420: Single Way ICache leaks uninitialized SeqMem contents

colinschmidt, 2016-10-25:
Due to a special case in Mux1H a single way Icache results in the datablock being set to the value coming out of the uninitialized data_array, despite no tag hits, where the multi-way case returns a literal zero.
Memories that represent non-initialized locations with X's will leaks these X's into the rest of the simulation.

@aswaterman Is this something you think should be fixed in upstream rocket-chip or should this be a backend specific change for my downstream branch?

https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/rocket/icache.scala#L135 for reference

aswaterman, 2016-10-25:
Finally we have arrived at issue #420blazeit

The direct-mapped I$ isn't the only place this will happen, unfortunately.  If you use a data scratchpad and don't use software to initialize the contents, this will happen, too.  The branch history table would have the same problem if mapped to SRAM.  We wrap our SRAM models to turn X into $random to work around this sort of thing... it seems to be the only general solution.

colinschmidt, 2016-10-25:
Alright thats what I thought just wanted to double check.

closed 2016-10-25 by colinschmidt