chipsalliance/rocket-chip #425: unresolved dependency when updating chisel

sashimi-yzh, 2016-10-28:
After applying patch 2b65478f3, I get the following error when generating verilog
```
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/yuzihao/riscv_pard_fpga/rocket-chip/sbt-launch.jar "run-main rocketchip.Generator /home/yuzihao/riscv_pard_fpga/generate rocketchip TestHarness rocketchip PARDFPGAConfig"
[info] Loading project definition from /home/yuzihao/riscv_pard_fpga/rocket-chip/project
Using addons: 
[info] Set current project to rocketchip (in build file:/home/yuzihao/riscv_pard_fpga/rocket-chip/)
[info] Updating {file:/home/yuzihao/riscv_pard_fpga/rocket-chip/}chisel...
[info] Resolving edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT ...
[warn]  module not found: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT
[warn] ==== local: tried
[warn]   /home/yuzihao/.ivy2/local/edu.berkeley.cs/firrtl_2.11/1.1-SNAPSHOT/ivys/ivy.xml
[warn] ==== public: tried
[warn]   https://repo1.maven.org/maven2/edu/berkeley/cs/firrtl_2.11/1.1-SNAPSHOT/firrtl_2.11-1.1-SNAPSHOT.pom
[warn] ==== sonatype-snapshots: tried
[warn]   https://oss.sonatype.org/content/repositories/snapshots/edu/berkeley/cs/firrtl_2.11/1.1-SNAPSHOT/firrtl_2.11-1.1-SNAPSHOT.pom
[warn] ==== sonatype-releases: tried
[warn]   https://oss.sonatype.org/content/repositories/releases/edu/berkeley/cs/firrtl_2.11/1.1-SNAPSHOT/firrtl_2.11-1.1-SNAPSHOT.pom
[info] Resolving jline#jline;2.12.1 ...
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  ::          UNRESOLVED DEPENDENCIES         ::
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn]  :: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
[warn]  ::::::::::::::::::::::::::::::::::::::::::::::
[warn] 
[warn]  Note: Unresolved dependencies path:
[warn]          edu.berkeley.cs:firrtl_2.11:1.1-SNAPSHOT
[warn]            +- edu.berkeley.cs:chisel3_2.11:3.1-SNAPSHOT
sbt.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
        at sbt.IvyActions$.sbt$IvyActions$$resolve(IvyActions.scala:313)
        at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:191)
        at sbt.IvyActions$$anonfun$updateEither$1.apply(IvyActions.scala:168)
        at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:156)
        at sbt.IvySbt$Module$$anonfun$withModule$1.apply(Ivy.scala:156)
        at sbt.IvySbt$$anonfun$withIvy$1.apply(Ivy.scala:133)
        at sbt.IvySbt.sbt$IvySbt$$action$1(Ivy.scala:57)
        at sbt.IvySbt$$anon$4.call(Ivy.scala:65)
        at xsbt.boot.Locks$GlobalLock.withChannel$1(Locks.scala:93)
        at xsbt.boot.Locks$GlobalLock.xsbt$boot$Locks$GlobalLock$$withChannelRetries$1(Locks.scala:78)
        at xsbt.boot.Locks$GlobalLock$$anonfun$withFileLock$1.apply(Locks.scala:97)
        at xsbt.boot.Using$.withResource(Using.scala:10)
        at xsbt.boot.Using$.apply(Using.scala:9)
        at xsbt.boot.Locks$GlobalLock.ignoringDeadlockAvoided(Locks.scala:58)
[error] (chisel/*:update) sbt.ResolveException: unresolved dependency: edu.berkeley.cs#firrtl_2.11;1.1-SNAPSHOT: not found
[error] Total time: 5 s, completed Oct 28, 2016 5:46:40 PM
make: *** [/home/yuzihao/riscv_pard_fpga/generate/rocketchip.PARDFPGAConfig.fir] Error 1
```

I find that all the URLs sbt tries do not exist.

aswaterman, 2016-10-28:
This hack https://github.com/ucb-bar/rocket-chip/blob/master/Makefrag#L27 is supposed to resolve the firrtl dependence by copying the jar into chisel3.  The rule triggers on any firrtl source change, which should have happened if you updated the firrtl submodule.  Can you investigate why the rule didn't fire?

sashimi-yzh, 2016-10-28:
I clean all files under `vsim/generated-src` and make again, it works.

Before the cleaning, I find that this target https://github.com/ucb-bar/rocket-chip/blob/master/vsim/Makefrag-verilog#L11 is triggered before building `$(FIRRTL_JAR)`. But I am confused about why it works after cleaning `generated-src`...

Anyway, I can not reproduce the problem now...

aswaterman, 2016-10-28:
> rocket-chip/blob/master/vsim/Makefrag-verilog#L11 is triggered before

Ah, that does look like a bug.  Anything that invokes sbt needs to depend
on FIRRTL_JAR.  I will fix it.

PR #426 (2016-10-28): Make all Chisel invocations depend on FIRRTL_JAR
Resolves #425
closed 2016-10-28 by aswaterman