chipsalliance/rocket-chip #429: assertions do not work in emulator

sashimi-yzh, 2016-10-29:
I add some features to the rocketchip, and I try to do some testings with `assert` statements in the scala code. First I try to make the emulator throw an assertion by adding the following code.
```
diff --git a/src/main/scala/rocket/icache.scala b/src/main/scala/rocket/icache.scala
index ef90453..2f5c167 100644
--- a/src/main/scala/rocket/icache.scala
+++ b/src/main/scala/rocket/icache.scala
@@ -137,6 +137,7 @@ class ICache(latency: Int)(implicit p: Parameters) extends CoreModule()(p) with
   }
   io.mem.acquire.valid := state === s_request && !io.s2_kill
   io.mem.acquire.bits := GetBlock(addr_block = refill_addr >> blockOffBits)
+  assert(io.mem.acquire.valid === Bool(true))
 
   // control state machine
   switch (state) {
```

I expect this will make the emulator fail on every tests, since the valid signal should not be always asserted. But all tests still pass, without throwing any assertions.

Do I use the `assert` statement wrong?

songmaotian, 2016-10-31:
assertion is scala runtime function, while emulator is executing the code generated by chisel, so I think it will not work, you have to find out how to generate stop signal in emulator

ccelio, 2016-10-31:
@sashimi-yzh , did you look at the output/*.out files? There could be a bug in the Makefile system recognizing a failure occurred.

@songmaotian I believe that Chisel may be changing Scala's "assert" to "require" and allowing Chisel to use "assert" as a Chisel node that can be tripped during run-time.

sashimi-yzh, 2016-10-31:
@ccelio I looked at them. I used `grep` to find `assert` and `fail` in all *.out files, and I found nothing. By using `tail -n 1 *.out`, all of them output `Completed after xxx cycles`. This may indicate that all tests pass.

zhemao, 2016-10-31:
@sashimi-yzh What happens when you just put in `assert(Bool(false))`?

sashimi-yzh, 2016-11-01:
I remove the `emulator/generated-src` directory and try again, and all things work as expected. Running the emulator actually throws assertions.

The `generated-src` directory before the removal contains results of other `$(PROJECT)` and `$(CONFIG)`. I am not sure whether they will affect the Makefile dependency. I try to reproduce the issue, but fail.

songmaotian, 2016-11-01:
@ccelio thx, I just learned that

terpstra, 2016-11-23:
Can this issue be closed?

sashimi-yzh, 2016-11-23:
I think it is ok to close.

closed 2016-11-24 by terpstra