chipsalliance/rocket-chip #430: TinyConfig Fails Some AMO Tests

palmer-dabbelt, 2016-10-31:
I was using TinyConfig to debug post-synthesis simulations and found that it was failing some tests.  It turns out these are failing upstream for TinyConfig.  They're not run on Travis, so I don't know how long they've been broken for.

````
  [ FAILED ] output/rv32ua-p-amoadd_w.out        (tohost = 669)  
  [ FAILED ] output/rv32ua-p-amoadd_w.out        (exit code =        669)  
  [ FAILED ] output/rv32ua-p-amoand_w.out        (tohost = 669)  
  [ FAILED ] output/rv32ua-p-amoand_w.out        (exit code =        669)  
  [ FAILED ] output/rv32ua-p-amoor_w.out         (tohost = 669)  
  [ FAILED ] output/rv32ua-p-amoor_w.out         (exit code =        669)  
  [ FAILED ] output/rv32ua-p-amoxor_w.out        (tohost = 669)  
  [ FAILED ] output/rv32ua-p-amoxor_w.out        (exit code =        669)  
  [ FAILED ] output/rv32ua-p-amoswap_w.out       (tohost = 669)  
  [ FAILED ] output/rv32ua-p-amoswap_w.out       (exit code =        669)  
````

aswaterman, 2016-10-31:
They don't fit in the 16 KiB scratchpad.  They could be rewritten to fit, but that would eliminate the microarchitectural test of AMOs that miss in the cache.

palmer-dabbelt, 2016-10-31:
OK, so I think it would be best if they could be dropped from the test list.  Is that something you'd be amenable to doing upstream, or should I do it myself?

aswaterman, 2016-10-31:
Dropping them isn't the right answer, either... I'll figure out a solution.

aswaterman, 2016-10-31:
I actually think the answer is to rely on torture to catch the microarchitectural bugs, and change these to be simple functional tests (and so they will work on TinyConfig).

ccelio, 2016-10-31:
That seems... unfortunate. Torture is a time-consuming pain to debug a failing case with, whereas the riscv-tests will throw an error code and you can jump to the source code +/- 5 lines of code with a comment that says "#try again after a cache miss".

I feel there needs to be a couple different levels of testing -- say cores can opt in (or even provide their own) tests to provide a more gradual rise in difficulty.

aswaterman, 2016-10-31:
Directed microarchitectural tests are a good idea, but should probably be separate from the ISA functionality tests.

sequencer, 2022-04-08:
Mark this issue depends on the new test framework to https://github.com/chipsalliance/rocket-chip/issues/2953

michael-etzkorn, 2022-04-12:
@sequencer I would say we just need documentation that TinyConfig won't pass riscv-tests that need more than 16kb of memory.

sequencer, 2022-04-13:
yeah we can document it firstly.