chipsalliance/rocket-chip #434: Berkeley Rocket-chip untethered implementation: available or not?

MarcoGeuna, 2016-11-07:
Hi all,

is there available an official untethered implementation of the Rocket-chip, made by UC Berkeley Architecture Research? I'm NOT referring to the one by lowRISC!

Among the closed issues, I read one from _ccelio_ [on Feb 22](https://github.com/ucb-bar/rocket-chip/issues/53) who wrote about a work-in-progress Berkeley untethered implementation, but on this repo I don't see any news about a new untethered Rocket-chip, and also in the [README](https://github.com/ucb-bar/fpga-zynq/blob/master/README.md) of the fpga-zynq repository it is only described a tethered version.

Taking into account these things, I suppose it is not available yet..isn't it? Please correct me if I'm wrong

Thank you!

ben-k, 2016-11-08:
I believe the master branch of rocket-chip is currently untethered.  The version in fpga-zynq has been modified to include a tether.

MarcoGeuna, 2016-11-08:
Thank you @ben-k for your answer!
So, my next question is: where can I find a new BBL? Because I can only find one [here](https://github.com/riscv/riscv-pk), but it is declared to be _"for tethered RISC-V systems"_. Then, is there available a tutorial to boot an untethered version, like the one you have for [fpga-zynq repository](https://github.com/ucb-bar/fpga-zynq/blob/master/README.md) (which is only for tethered)?
Thank you again!

zhemao, 2016-11-08:
Rocket-chip master is tethered, as you have to use the debug interface in order to get anything done. If you have some storage device, you can add it as a peripheral and modify the bootloader to load the second-stage bootloader off of that instead of waiting for a program to be loaded by the tether.

We currently do not have any official open source releases capable of standalone boot.

BBL shouldn't require a tether in order to boot the OS. But it will need the tether in order to print to a serial console.

MarcoGeuna, 2016-11-13:
Ok! Got it... Thank you @zhemao for your answer!

closed 2016-11-16 by zhemao