chipsalliance/rocket-chip #436: Cannot generate an instance of rocket-chip with 0 top-level interrupts

davidbiancolin, 2016-11-16:
In fpga-zynq, we use a configuration of rocket-chip that presents no external interrupts at the top level. 

Omitting PeripheryExtInterrupts{ , Bundle, Module} in the ExampleTop cake (as we have done previously), or alternatively setting NExtInterrupts = 0 both produce exceptions. 

Given the trivial workaround, this is not critical but it does represent regression from commit 5392219 (the current head of submodule in fpga-zynq).

aswaterman, 2016-11-16:
agreed, there's no good reason not to support this case.

terpstra, 2016-11-16:
I'll look into this as soon as we're done the L1 conversion.

terpstra, 2016-11-19:
Fixed in commit 5b594ced293f323323611ca65de80b59d55ab959.

closed 2016-11-19 by terpstra