chipsalliance/rocket-chip #437: Cannot mix in traits that set pBusMasters > 0

davidbiancolin, 2016-11-16:
In fpga-zynq we use a serial adapter/tether to write into the L2 of rocketchip. 

When i mix the serial adapter traits into my cake i get an exception (below). The same error can also be exercised by setting NExtBusAXIChannels > 0 in the ExampleTop, which should be handled appropriately by the PeripherySlave cake mix in. 

```
[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
        at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
        at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
        at chisel3.core.Module$.do_apply(Module.scala:29)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:190)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:190)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:194)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:192)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.internal.Builder$.build(Builder.scala:192)
        at chisel3.Driver$.elaborate(Driver.scala:190)
Caused by: cde.ParameterUndefinedException: Parameter NCoreplexExtClients undefined.
        at cde.World$TopLookup$1.apply(Parameters.scala:156)
        at cde.World$_View.sym(Parameters.scala:114)
        at cde.Parameters$$anonfun$makeMask$3.apply(Parameters.scala:300)
        at cde.Parameters$$anonfun$makeMask$3.apply(Parameters.scala:295)
        at cde.Parameters$KidLookup$1.apply(Parameters.scala:337)
        at cde.World$_View.apply(Parameters.scala:111)
        at cde.View.apply(Parameters.scala:38)
        at coreplex.BaseCoreplexConfig$$anonfun$1.apply(Configs.scala:118)
        at coreplex.BaseCoreplexConfig$$anonfun$1.apply(Configs.scala:20)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:23)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:22)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:23)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:22)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:23)
Caused by: scala.MatchError: NCoreplexExtClients (of class rocketchip.NCoreplexExtClients$)
        at cde.Config$$anonfun$$lessinit$greater$default$1$1.apply(Config.scala:4)
        at cde.Config$$anonfun$$lessinit$greater$default$1$1.apply(Config.scala:4)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:26)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:22)
        at cde.World$TopLookup$1.apply(Parameters.scala:156)
        at cde.World$_View.sym(Parameters.scala:114)
        at cde.Parameters$$anonfun$makeMask$3.apply(Parameters.scala:300)
        at cde.Parameters$$anonfun$makeMask$3.apply(Parameters.scala:295)
        at cde.Parameters$KidLookup$1.apply(Parameters.scala:337)
        at cde.World$_View.apply(Parameters.scala:111)
        at cde.View.apply(Parameters.scala:38)
        at coreplex.BaseCoreplexConfig$$anonfun$1.apply(Configs.scala:118)
        at coreplex.BaseCoreplexConfig$$anonfun$1.apply(Configs.scala:20)
        at cde.Config$$anonfun$addDefinitions$1.apply(Config.scala:23)
java.lang.RuntimeException: Nonzero exit code: 1
        at scala.sys.package$.error(package.scala:27)
        at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
        at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2077)
        at scala.Option.foreach(Option.scala:236)
        at sbt.BuildCommon$class.toError(Defaults.scala:2077)
        at sbt.Defaults$.toError(Defaults.scala:39)
        at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
        at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
        at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
        at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
        at sbt.std.Transform$$anon$4.work(System.scala:63)
        at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
        at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
        at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)

```

Unlike the previous bug, this one i cannot workaround so easily, nor can I make sense of what is happening from the stack trace. As far as i can tell NCoreplexExtClients is being correctly calculated and set, at least initially. This only manifests near the end of elaboration.

sashimi-yzh, 2016-11-16:
I meet this issue, too. I want to add a DMA client to the rocketchip, so I set `NExtBusAXIChannels` to `1`. The result gives me the same exception.

terpstra, 2016-11-16:
Hi. I am not terribly surprised by this exception, as those NExtBusAXIChannels is 1-2 days away from being removed. There will be instead a new example of how to hook up additional AXI masters to the updated L2. As soon as the L1 changes land, we will be re-adding AXI mastering as we also need this for a very near deadline!

terpstra, 2016-11-16:
PS. It will also be much faster. At least 4* improvement in bandwidth. Possibly more.

terpstra, 2016-11-19:
Fixed in commit 5b594ced293f323323611ca65de80b59d55ab959.

terpstra, 2016-11-19:
Argh. Wrong issue!

terpstra, 2016-11-23:
Fixed in #447.
See https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/rocketchip/Periphery.scala#L157 (PeripherySlaveAXI4) for an example of how to create an AXI4 master port.

closed 2016-11-23 by terpstra