chipsalliance/rocket-chip #463: .cfg not emitting `ram` section (PK breakage)?

seldridge, 2016-11-29:
Off of master (4234cff0744bae1f602b84011dbef261584c0a27), I'm getting a PK failure with [`query_mem`](https://github.com/riscv/riscv-pk/blob/master/machine/configstring.c#L10) failing to find `ram{0{addr` in the config string.

Looking at an old config string, I'm seeing `ExampleRocketTop` emitted twice, once with what the `ram{0{` region used to have and once with it's old contents. The new config string looks like:
```
ExampleRocketTop {
  addr 0x80000000;
  size 0x10000000;
}
...
ExampleRocketTop {
  addr 0x60000000;
  size 0x20000000;
}
```

The old config string looks like:
```
ram {
  0 {
    addr 0x80000000;
    size 0x10000000;
  };
};
...
ExampleTop {
  addr 0x60000000;
  size 0x20000000;
}
```

terpstra, 2016-12-01:
The heart of the problem is that we have no specification for what cfg string should contain and how it should be organised. We only have a syntactical format. Once we define what should be in a config string, then we can define what's right/wrong for a given device. Hopefully this will get sorted out soonish, so that we can then output the right thing. Until then, I don't see the point in massaging the config string output.

seldridge, 2016-12-01:
Seems fine to me. You can always kludge around it in the proxy kernel if absolutely needed ([hardcoding what should be read in `query_mem`](https://github.com/ucb-bar/rocket-chip/issues/467#issuecomment-263801175)).

Thanks for looking into this and for the explanation.

closed 2016-12-01 by hcook

terpstra, 2016-12-01:
Superseded by #474