chipsalliance/rocket-chip #466: Should we make ExtMem and ExtBus more easily configurable?

hcook, 2016-11-29:
Currently BasePlatformConfig includes MasterConfig fields for ExtMem and ExtBus, but we don't provide helper WithConfigs for overriding them. Should we?

hcook, 2016-11-29:
The specific user feature request was to override the base address of ExtMem

zhemao, 2016-11-30:
It definitely makes sense to make the external memory size configurable. Then it can be adjusted to match the size of the actual backing memory.

terpstra, 2016-11-30:
My main concern is that we will need multiple fragments of memory for more
than 2GB machines. So the right answer on how to generealize membase is not
just adding a with config.

terpstra, 2017-02-17:
@hcook You and other Henry have some patchset that fixes this, right?

davidbiancolin, 2017-03-06:
For what it's worth, we're interested in configurations with >2 GB here at Berkeley.

closed 2017-03-24 by hcook