chipsalliance/rocket-chip #470: Travis fails all rocketsuites recently

colinschmidt, 2016-11-30:
Does anyone know why all of the rocket regression suites have been timing out recently?
Is this just a travis update that we have to deal with?

hcook, 2016-11-30:
I'm looking into this too but haven't made progress yet. Build #2113 passed and build #2114 failed, which strikes me as very suspicious considering #2114 is just a merge commit...

hcook, 2016-11-30:
Then UnitTestSuite passes in #2115 before failing again in #2116; none of that code has been touched.

colinschmidt, 2016-11-30:
Yeah thats why I was suspecting some weird travis change on their side but based on their twitter and update schedule page it doesn't seem like anything should have changed.

Thanks for looking into it.

hcook, 2016-11-30:
Comparing 2115.6 and 2116.6 (UnitTestSuite passing then failing) it looks like the biggest difference is that "make tools verilator -C regression SUITE=none" [install] takes  0.67s vs 2179.47s!

hcook, 2016-11-30:
https://www.traviscistatus.com/incidents/nq0sf0srvx8s has been going on for about the right timeframe, but they claim only affects OSX builds...

hcook, 2016-11-30:
I was thinking it could be because bumping riscv-tools invalidated the travis cache, but that doesn't explain why #2113 passed and #2114 failed...

hcook, 2016-11-30:
Looking at https://travis-ci.org/ucb-bar/rocket-chip/caches you can see the caches of the failing builds are notable smaller than the successful ones, but I'm not sure why different jobs are using/creating different caches... https://docs.travis-ci.com/user/caching/#Caches-and-build-matrices implies that jobs using the same cache can corrupt it, but I'm not sure how that's applicable here

terpstra, 2016-12-01:
I looked into this a while ago. So let me explain some of the things I know:
1. The size of the cache you see on the /caching URL is 600MB * # of travis suites. So it is bigger when all the suites pass.
2. When there is a cache, then install takes 0.67s, otherwise it is super slow.
3. The build time for riscv-tools went up considerably recently.
4. In order for Travis to work, both the uncached install + the suite test must fit in 50 minutes.
5. I was unable to merge a PR that bumped all the submodules earlier, because it caused uncached rebuilds to exceed the 50 minute target. You guys managed to merge a bump somehow anyway.

hcook, 2016-12-01:
I'm looking into using container images stored on DockerHub to cache pre-built toolchains

zhemao, 2016-12-07:
This has been fixed, right?

terpstra, 2016-12-07:
Not completely

hcook, 2017-01-11:
Using docker images instead of the travis cache got us back under the time limit, though average time is up.

Any bump to riscv-tools now requires a docker image to be re-built, and this is not automated. Ping me if you bump riscv-tools. Hopefully upstreaming will resolve some of this headache in the future.

closed 2017-01-11 by hcook