chipsalliance/rocket-chip #473: Rocket should trap on unsupported AMOs

terpstra, 2016-12-01:
The TLB is in a position to know which memory regions support Arithmetic and Logical atomic operations. Currently, the processor will just issue them if the memory region is RW. That means you can violate the invariant of the TileLink network that only legal operations are inflight. The processor should probably check if an AMO is supported before it is issued.

It is probably reasonable to just require() that if a device supports any AMOs as seen by rocket, it supports 32 & Xen sized AMOs. That way the runtime check boils down to a simple Address=>Bool function.

terpstra, 2017-03-26:
@aswaterman Has your new TLB resolved this issue?

aswaterman, 2017-03-26:
Not yet, I'll close this issue once it's addressed.

terpstra, 2017-04-15:
@aswaterman This was solved in https://github.com/ucb-bar/rocket-chip/pull/680, right?

closed 2017-04-15 by terpstra

aswaterman, 2017-04-15:
yeah, thanks for closing