chipsalliance/rocket-chip #475: Document Configs/Parameters

mwachs5, 2016-12-01:
I've been getting a lot of questions about "what do all the Configs do"?

Is there a way we should document the Configs/Parameters, or is it just RTFC? If it's just RTFC we should at least comment what each one does...

Should we create a table? Or just comment each thing a bit more?

richardxia, 2016-12-01:
I think if we document it, it should be done via scaladoc comments, which could be used to generate HTML docs.

edcote, 2017-04-12:
Wouldn't it be possible to pull all configs via scala's reflection API and make configuration self documenting?

sequencer, 2022-04-08:
related to https://github.com/chipsalliance/rocket-chip/issues/2960
closing for now to clean up the issue list.

closed 2022-04-08 by sequencer