chipsalliance/rocket-chip #477: Processor diagram

LyronDev, 2016-12-05:
I've been having trouble understanding the rocket chips architecture and pipeline stages in the chisel code.
Is there a visual representation like there is for the Sodor Processor Collection (see [here]) ?((https://github.com/ucb-bar/riscv-sodor/blob/master/doc/2stage.pdf))

ben-k, 2016-12-05:
Perhaps you're looking for something like this:

http://www.lowrisc.org/docs/tagged-memory-v0.1/rocket-core/

Be careful, though; that picture was made more than a year ago, so it may no longer be up to date.  I'm not aware of a more recent version with that level of detail.

LyronDev, 2016-12-05:
That is indeed what i'm looking for. How did i not find this, urgh..
Thank you, i'll look into it

closed 2016-12-05 by ben-k