chipsalliance/rocket-chip #483: General Emulation Performance Slowdown vs. Chisel2

seldridge, 2016-12-07:
I'm seeing a relatively dramatic slowdown (25x--100x) compared to the old Chisel2 version of Rocket Chip. This is getting hit from both the Chisel3 version running a lot longer (10x the cycles) and the Chisel3 version running a lot slower (1/5 the emulation frequency). Note, _there are a lot of moving pieces here_, Chisel3/FIRRTL/Verilator/Proxy Kernel/toolchain, etc. etc.

Note: the Chisel3 version has 5x as many lines of Verilog than the Chisel2 version, so this is a markedly "bigger" thing to emulate.

### With the Proxy Kernel (~25x)
Test program (`do-nothing.c`):
```c
int main(int argc, char ** argv) {}
```

Chisel2 (74KHz for 0.3 million cycles on  c4c6bd1 from April 1, 2016):
```
> time ./emulator-Top-DefaultCPPConfig +cycle-count pk /tmp/do-nothing
Completed after 370707 cycles

real	0m5.083s
user	0m3.552s
sys	0m0.004s
```

Chisel3 (16KHz for 2.2 million cycles on 915697c from December 6, 2016):
```
> time ./emulator-rocketchip-DefaultConfig +cycle-count pk /tmp/do-nothing
Completed after 2270159 cycles

real	2m22.079s
user	2m18.412s
sys	0m0.156s
```

### Bare Metal (~100x)
Chisel2 (84KHz for 12k cycles):
```
> time ./emulator-Top-DefaultCPPConfig +cycle-count /opt/riscv/stow/riscv-tools-2016-03-14-34a245b/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
Completed after 12645 cycles

real	0m0.150s
user	0m0.124s
sys	0m0.000s
```

Chisel3 (12KHz for 171k cycles):
```
> time ./emulator-rocketchip-DefaultConfig +cycle-count /opt/riscv/stow/riscv-tools-2016-11-21-ad9ebb8/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
Completed after 171270 cycles

real	0m14.483s
user	0m13.660s
sys	0m0.088s
```

I believe you all were doing some evaluation of the performance ([from a brief discussion on the mailing list](https://groups.google.com/forum/#!searchin/chisel-users/verilator%7Csort:relevance/chisel-users/MypZWRrNvTk/O8e0xMMGMgAJ)). Has there been any further discussion internally?

ccelio, 2016-12-07:
At least one confounding aspect is the removal of HTIF and the new addition of the Debug Module (which I believe answers why the programs are running for a lot longer). 

Before, proxying syscalls was handled by the rocket core poking the fesvr via tohost and sharing information via HTIF reading target memory. That's been replaced by the Debug Module (DTM) polling the rocket-chip very frequently and having code run on the rocket core itself to answer the question "do you have any syscalls or data you'd like fesvr to handle?"  The bandwidth to get memory off the target via the DTM is very low. Also, programs are loaded via the DTM, which is also a painful, slow process. "rv64ui-p-simple" is now ~3k instructions, I believe.

While that won't answer why the target simulation frequency is degraded, it does help answer "why is my dhrystone score so bad?"

kammoh, 2016-12-07:
I should also voice out my concern about the simulation slowdown here
(close to 80x in running dhrysone). Is there proof that none of this could
as well be attributed to the verilator simulation? Are there any plans for
simulation speedup? What are the steps we can start working on to get back
to the previous performance figures?

p.s: The old KIPS figures were really good and somewhat analogous to
dominant fullsystem simulators. That could have opened the possibility of
using the model in place of a fs simulation engine, while magically having
access to all the implementation details and staying "synced" to the RTL.

seldridge, 2016-12-07:
Thanks for that info @ccelio. That would seem to go a long way towards explaining the magical extra cycles. Maybe the Debug Module could be configured to back off for some time? That would certainly seem to help for the bare metal tests.

I don't know what's going on with Verilator, but it's possible that it is being efficient (I am skeptical, though). The Chisel3 version does have ~5x the number of lines of Verilog and I don't know how the old Chisel2 C++ backend would've performed given an equivalent number of statements. It's possible that Verilator is just getting tied up with 5x as many Verilog statements that it can't optimize. That coupled with a 5--10x increase in the cycles due to the Debug Module may put us where we are (>25x slower).

@kammoh, yes, the old KIPS was great. I just got used to that... The good news is that the performance degradation is so bad that I'd anticipate that there exist low-effort approaches to recover some of that performance.

kammoh, 2016-12-07:
Does anyone have the the current KIPS figures for simulation under VCS that
we can compare to?

mwachs5, 2016-12-07:
> Why is fwsrv

Has anyone considered changing fesrvr so that it uses a more efficient
polling mechanism with the debugger? If the hart wrote to debug memory or
halted when it wanted to talk to the debugger (instead of writing an
arbitrary memory address) the simulated debugger could poll with no impact
on the running code.

Megan

colinschmidt, 2016-12-07:
VCS is 2x faster than verilator for me on a rocket-chip with hwacha.

mwachs5, 2016-12-07:
Also, are we comparing apples to oranges in the lines of verilog? At the same time we switched from chisel2 to chise3 we also added lots of TL monitors. These add a lot of lines of code without impacting the "real" performance (but could certainly impact simulation performance)

seldridge, 2016-12-07:
@mwachs5, possibly. Maybe you can see if I'm thinking about this right... It would seem that a naive cycle-based simulator is going to scale linearly in runtime with additional lines of Verilog (assuming a line of Verilog is some simple assignment like what FIRRTL emits). Unless it's doing something intelligent, it's going to just do dumb clock_lo/clock_hi sequential/combinational updates. However, I don't recall the Chisel2 C++ backend doing anything particularly intelligent or trying to be smart about not executing specific module updates. This leads me to believe that Verilator is just dropping performance (but the Chisel2 vs. Chisel3 comparison is really just another apples/oranges thing beyond the lines of Verilog argument). I would expect an event driven simulator (e.g., VCS) to scale with the activity of the design, hence all the new TL stuff shouldn't matter. Some [old slides about Verilator](http://www.veripool.org/papers/Verilator_Fast_Free_Me_DVClub10_pres.pdf) indicate that Verilator's optimizations are not exotic (e.g., no skipping module updates).

@colinschmidt, thanks for that datapoint. Using VCS or some other simulator is an option for us -- I'd still hope that more of the 5x in simulation frequency could be recovered. For reference, do you happen to remember the ballpark VCS performance vs. the Chisel2 codebase?

FWIW, part of my Verilator skepticism [claims anecdotally that it is both __1.5x faster than VCS__ and that __VCS is 3x faster__](http://www.veripool.org/wiki/veripool/verilog_simulator_benchmarks)...

colinschmidt, 2016-12-07:
So it turns out when we initially were making the transition I did do some benchmarking with hwacha. This was a while ago and I couldn't tell you the exact commit but at that point it was a ~20% slow down on VCS moving from Chisel2 to Chisel3.

zhemao, 2016-12-07:
@mwachs5 We have sort of solved this problem in fpga-zynq and project-template by having a serial tether that can read the tohost/fromhost values directly from memory. This at least avoids the performance degradation from constantly interrupting the core.

I will say that, anecdotally, long-running simulations (like Linux kernel boot) go much faster with verilator than with VCS.

mwachs5, 2016-12-07:
We should just (optionally) implement the serial interface option of the Debug Module. Then we can all use the same code base

colinschmidt, 2016-12-07:
My numbers of simulation runtime were for a large suite of bare-metal assembly tests, not a single long running benchmark. So perhaps verilator just has a high startup cost?

aswaterman, 2016-12-07:
The interface is a red herring.  The performance degradation isn't coming from using TSI vs. DM as transport; it's that the DM doesn't implement the memory master port, whereas TSI requires one.

For systems that have a coherence system, we could conditionally implement the DM's master port, and kill TSI.

mwachs5, 2016-12-07:
Yeah @aswaterman  I guess in my previous comment I meant "Bus Master" not "Serial". Though that brings up lots of questions about coherence, etc.

aswaterman, 2016-12-07:
Makes sense - and we can fall back to the intrusive mechanism when there's no coherence in HW.

zhemao, 2016-12-07:
@aswaterman do you mean that, instead of writing just to the debug RAM, the debug transport could also read/write to arbitrary parts of the address space?

mwachs5, 2016-12-07:
This is part of the Debug Spec -- optional "Bus Master"

sequencer, 2022-04-08:
Since Chisel2 is dropped a long time ago, closing this issue to clean up the issue list.
But we can benchmark `SFC` vs `MFC` on both `VCS` and `Verilator` after we have new test framework.

closed 2022-04-08 by sequencer