chipsalliance/rocket-chip #484: Better emulator.cc command line argument processing (i.e., getopt_long)

seldridge, 2016-12-08:
I've never been a fan of the way that [`emulator.cc`](https://github.com/ucb-bar/rocket-chip/blob/master/csrc/emulator.cc#L48) does command line processing. There's no help text, the normal version will happily read options that it has no business understanding (the user can specify a VCD file and a start cycles to the non-debug emulator), the command line processing is bespoke, and all the `+` options just seem like an odd choice. I'd like to propose cleaning this up with the following depending on what will get merged:
  * Add help text to `emulator.cc`
  * Have the non-debug version error out if passed command line arguments that it can't handle
  * Have the emulator dump help text if the user doesn't specify at least one non-command line argument
  * Refactor `emulator.cc` to use `getopt_long`

The first three are trivial, but if I do the latter, the cleanest way is to convert all existing `+` arguments to short or long `-` equivalents and just rely on `getopt_long` to deal with the processing. This is a pretty big (superficial) change to the longstanding way of running the emulator and has some far reaching (though again, superficial) consequences to all the Makefiles that call the emulator. This similarly has the potential to break build flows for people that may be using these arguments.

I can get together a pull request (in my free time), but what here sounds good enough to merge?

seldridge, 2016-12-08:
Okay, there's more interactions here than expected. The real way that the emulator should be processing the options is dependent on what htif_t expects. The first command line argument that doesn't begin with '-' or '+' is assumed to be the start of whatever program is going to be run. There's an underlying problem here where `emulator.cc` will use arguments to the binary that it shouldn't. I used to run into this all the time where I'd have a `-v` flag for some program run through the proxy kernel and the emulator would try to consume it.

I'll align `emulator.cc` with how hitf_t expects the command line arguments to look. This then should be done without needing to change anything (e.g., Makefiles).

aswaterman, 2016-12-08:
Hold up, doesn't getopt_long infect with GPL?

seldridge, 2016-12-08:
I am referring to the glibc version here, however, there does exist a [bsd variant](https://github.com/freebsd/freebsd/blob/386ddae58459341ec567604707805814a2128a57/crypto/openssh/openbsd-compat/getopt.h). Is it enough that you don't specify which version is being linked against or is this something that should just be avoided?

aswaterman, 2016-12-08:
Conceivably, we could snag the BSD version into rocket-chip/csrc and unconditionally build that.

Alternatively, we could use fesvr/option_parser.h, which is more c++ friendly and is used by spike.

seldridge, 2016-12-11:
Thanks for the ideas @aswaterman. I don't think I can ram what I think needs to be in emulator.cc using fesvr's option parser without modifying this to properly deal with '+' args. I'd also like to actually keep the '+' args available as removing these would break an existing API for really no benefit.

FWIW, users building this may be pulling in GPL headers with the current state of `emulator.cc`. This uses, e.g., `stdio.h`, which for me (RHEL6) is defaulting to an LGPLv2 version. I do have a BSD version of `stdio.h` installed, but I'm not using it. I expect that this would be the case for the majority of rocket-chip users. My "default" version of `getopt.h` is the same license. I can obviously imply what I mean by doing an `#include bsd/getopt.h`, but I think if that's an issue, you need to be doing that for all the standard libraries? I'm a little hazy on the licensing, but I certainly don't want to do anything that pushes this all to GPL unnecessarily.

#487 addresses this, but does not take care of the `getopt.h` question.

sorear, 2016-12-11:
I have never heard of this problem with getopt_long before.  glibc is glibc, so the license issues for getopt_long are the same as any other libc function… if we want to make the emulator run on platforms **without getopt_long**, then we would need to copy an implementation into our project, and then we would need to care about licensing.  But it's pretty explicit that linking against symbols from libc is not a GPL issue, otherwise there would be no proprietary software on glibc Linux…?

aswaterman, 2016-12-12:
I might have been confusing getopt_long with readline... anyway, LGPL's fine.

commit 7e794212 (2016-12-13): Cleanup emulator.cc, use getopt, add help text (#487)

This changes the emulator to conform to POSIX-style options (e.g.,
short/'-' and long/'--') while preserving legacy option parsing (i.e.,
'+'). Options are read from the user until the first
non-option (either POSIX or legacy) is encountered. This and
everything following is assumed to be the binary and any arguments the
user wants to run on the emulator. All non-options are passed directly
to the DTM. This allows for the same option to be passed, safely, to
both the emulator and the binary, e.g., "+verbose".

This introduces a dependency  [... cut]
closed 2016-12-13 by hcook