chipsalliance/rocket-chip #490: TinyConfig error

samiasd, 2016-12-13:
Dear all,
is it possible to run any benchmark on TinyConfig architecture??
 DefaultRV32 works fine, but when trying to execute benchmarks on Tiny I got the following error:

*** FAILED *** (tohost = 1337)
C         0:     581525 [0] pc=[00000838] W[r 0=80003ea0][0] R[r 0=00000000] R[r18=00000003] inst=[7b200073] DASM(7b200073)
*** FAILED *** (code = 1337, seed 1481632547) after 581526 cycles

thanks

aswaterman, 2016-12-13:
TinyConfig has a very small scratchpad, and no FPU, either of which can
cause the benchmarks to fail.

samiasd, 2016-12-14:
@aswaterman 
so it is not possible to test TinyConfig in any case?? what is the smallest processor that I can test it without problem??
thanks

aswaterman, 2016-12-15:
It is possible.  The default tests and benchmarks assume a larger amount of RAM, which assumes the luxury of a 

You could probably increase the size of the scratchpad to run those tests.

closed 2016-12-15 by aswaterman