chipsalliance/rocket-chip #495: wrong corrected data used in L1 NBCache's correction replay?

wsong83, 2016-12-16:
in src/main/scala/rocket/NBDcache.scala
around line 844, in the L1 NB Cache

s3_req.data := Mux(s2_data_correctable, s2_data_corrected, amoalu.io.out)

When s2_data_correctable, should it be
  s2_data_corrected >>  Cat(s2_word_idx, Bits(0,log2Up(coreDataBits)))
rather than
  s2_data_corrected
?

Since there is no real parity check in L1, this branch is never checked.
What I can see is s2_data_corrected is a flattened vector and the requested word does not always located at index 0 unless doNarrowRead.

aswaterman, 2016-12-16:
Not surprising; the ECC support in the L1 D$ is incomplete and so might have bugs.  @terpstra can you make sure this is the right fix?

Thanks for the bug report, @wsong83!

terpstra, 2016-12-17:
I think this would be better dealt with by @hcook, who last updated the code for the TL2.

sequencer, 2022-04-08:
close since #2053 merged.

closed 2022-04-08 by sequencer