chipsalliance/rocket-chip #497: CDE, a dependency of other ucb-bar repos, has been removed from rocket-chip

mward15, 2016-12-23:
For example, fpga-zynq's testchipip requires cde to be built. However, CDE has been excluded with https://github.com/ucb-bar/rocket-chip/pull/444/commits/e8e95d4bcf025f266070819c266c7d7ddae8676f commit, so now fpga-zynq could be built only with October version of rocket-chip repository (5392219d86bd8ff3b241122bd38852e89fa06ac2) More details could be found at this resolved **fpga-zynq** issue - https://github.com/ucb-bar/fpga-zynq/issues/32 . Simply downloading the CDE repository (https://github.com/ucb-bar/context-dependent-environments) does not solve the issue, because the CDE functionality is being moved to "config", which results in a lot of conflicts between CDE and config while trying to build testchipip linked to CDE

This is a major problem - not just because it is sad to have to stay on the outdated version, but also because at this specific October version of rocket-chip: the emulator (make debug) and riscv-gnu-toolchain builds are failing - meanwhile, at the recent December version of rocket-chip they build OK

Please tell, is it possible to somehow locally combine the downloaded October and December versions of rocket-chip, so that all three things (testchipip, emulator and riscv-gnu-toolchain) could be compiled? Or that could give the compatibility problems? (e.g. verilog fails to build)

aswaterman, 2016-12-23:
rocket-chip is not a compatibility layer for CDE or fpga-zynq.  You should ask the maintainers of fpga-zynq for advice on this matter.  Good luck!

closed 2016-12-23 by aswaterman

davidbiancolin, 2016-12-23:
@mward15 Feel free to open another issue on fpga-zynq and I can direct you better from there.