chipsalliance/rocket-chip #500: Is ScalaDoc documentation available?

MarcoGeuna, 2016-12-30:
Hi all,

Do you have available some documentation that explains how your packages and their classes, traits etc are organized, and their methods? I'm thinking about something like ScalaDocs...I found only [this](https://chisel.eecs.berkeley.edu/api/#package) about Chisel, and [this](https://github.com/ucb-bar/uncore) about uncore package. What about all the other packages?

Thanks!

ben-k, 2017-01-09:
Alas, such documentation does not exist (though it would sure be nice!).  Also, be careful - the uncore submodule that you linked to is an old, deprecated version.

Hopefully there will be less churn in the codebase going forward and so comprehensive documentation will become feasible.

closed 2017-01-09 by ben-k

MarcoGeuna, 2017-01-10:
Ok! Thank you for your answer!