chipsalliance/rocket-chip #505: Reporting "'A' channel re-used a source ID" while running bbl during simulation

sashimi-yzh, 2017-01-11:
I have added the UART scala code from the sifive repo to rocketchip. But the simulation triggers an assertion fail about re-using source ID. Here is the simulation output (I have commented out the scala code printing rocket execution logs) :
```
using random seed 1484156486
line 137: hartid = 0
              vvAssertion failed: 'A' channel re-used a source ID (connected at Tile.scala:57:13)
    at Monitor.scala:418 assert(!inflight(bundle.a.bits.source), "'A' channel re-used a source ID" + extra)
```
The `line 137: hartid = 0` is a debugging message I insert in the bbl. [It is output by UART char by char](https://github.com/sifive/sifive-blocks/blob/master/src/main/scala/devices/uart/UART.scala#L79). The two letters `v` before the assertion message are parts of the RISCV logo. For reference, codes around line 57 in my `Tile.scala` are
```
 55   val cachedOut = TLOutputNode()
 56   val uncachedOut = TLOutputNode()
 57   cachedOut := dcache.node
 58   uncachedOut := tileXbar.node
 59   val masterNodes = List(cachedOut, uncachedOut)
```

But after I try to disable the TLmonitor with `++ new WithoutTLMonitors` in `src/main/scala/rocketchip/Configs.scala`, simulation can print the full logo without crashing, even boot linux and print "Hello world" over linux!

As a result, will it be possible that the assertions in TLmonitor are a little bit of over-strict?

hcook, 2017-01-11:
The monitor assertions are there to ensure compliance with the TileLink protocol specification; i.e. that masters and slaves implement the protocol correctly. While it's possible for non-compliant implementations to function, that doesn't mean that the assertions are too strict. Instead, I'd view this as a bug in the DCache master implementation. The DCaches were recently convert to use TileLink2 instead of TL1, so it's definitely possible there's a bug our tests didn't catch.

Which Config are you using, and specifically which kind of DCache is being instantiated?

sashimi-yzh, 2017-01-12:
I use the following Config:
```
class DefaultConfig extends Config(new WithBlockingL1
  ++ new WithoutFPU
  ++ new WithExtMemSize(0x4000000L)
  ++ new BaseConfig
)
```
I also use the MultiClockCoreplex.

I hope this help.

sashimi-yzh, 2017-01-20:
For reference, using MultiClockCoreplex with the original DefaultConfig to run asm tests will also trigger similar assertion fail.
```
Assertion failed: 'A' channel re-used a source ID (connected at HellaCache.scala:159:18)
    at Monitor.scala:418 assert(!inflight(bundle.a.bits.source), "'A' channel re-used a source ID" + extra)
```

terpstra, 2017-03-26:
I would like to see this issue resolved. Can you provide some step-by-step instructions that we could follow to reproduce this issue?

sashimi-yzh, 2017-03-27:
I check again. It is resolved! Now no assertion is reported during outputting the logo.

Instructions to reproduce this issue over today's master branch are so kind of complex. Here are the modifications
* modify the register space of UART provided by sifive-blocks to a Xilinx UARTLITE, since Xilinx UARTLITE is very simple and the driver is easy to write
* map the modified UART to the MMIO space of rocketchip
* modify bbl to access the register space UART to print chars, instead of calling the riscv-pk
* disconnect the debug ports of rocketchip in TestHarness
* write a DMA state machine to transfer the bbl from host disk to the memory of rocketchip. This can be done with the help of `$readmemh` in verilog

I think transferring bbl image in this way is more efficient than using the debug module. With the commit log disable in `src/main/scala/rocket/Rocket.scala`, it only cost about 22s to transferring a bbl image of size 68KB into the memory of rocketchip, and cost another 20s for rocket to run bbl and output the riscv logo. To boot linux with about 64MB memory and run a hello world program as init, it only cost about 20 minutes.

I do not know whether this way can help you to shorten the time to do testing, since it requires a lot of modifications over the master branch. If you think it helps, I can push the code for reference. :)

terpstra, 2017-03-27:
Well, if you say the problem is resolved, then I guess we do not need to reproduce it!

sashimi-yzh, 2017-03-28:
I find that a new priv-1.10 branch is merged into master just now. Let me check this issue again. :)

But it may cost some time, because I am preparing a presentation.

terpstra, 2017-04-15:
So... can we close this issue?

sashimi-yzh, 2017-04-15:
Oh, sorry, I forgot this. OK, everything is fine.

closed 2017-04-15 by terpstra