chipsalliance/rocket-chip #509: rocket-chip/riscv-tools/build.sh fails

mox-mox, 2017-01-13:
When I try to generate the the RISCV toolchain, the build.sh script fails with "undefined reference to `yywrap'".

System: Current Archlinux, 64 Bit

Steps to reproduce:
    git clone https://github.com/ucb-bar/rocket-chip.git
    cd rocket-chip
     git submodule update --init --recursive
    cd riscv-tools
    export RISCV=/some/folder/not/in/rocket-chip
    ./build.sh

This yields:
    Starting RISC-V Toolchain build process
    
    Configuring project riscv-fesvr
    Building project riscv-fesvr
    Installing project riscv-fesvr
    mkdir //tmp/asd-mox/home/mox/uni/s11/Seminar/rc_toolchain/include
    mkdir //tmp/asd-mox/home/mox/uni/s11/Seminar/rc_toolchain/include/fesvr
    mkdir //tmp/asd-mox/home/mox/uni/s11/Seminar/rc_toolchain/lib
    mkdir //tmp/asd-mox/home/mox/uni/s11/Seminar/rc_toolchain/bin
    mkdir //tmp/asd-mox/home/mox/uni/s11/Seminar/rc_toolchain/lib/pkgconfig
    
    Configuring project riscv-isa-sim
    Building project riscv-isa-sim
    Installing project riscv-isa-sim
    mkdir //tmp/asd-mox/home/mox/uni/s11/Seminar/rc_toolchain/include/spike
    
    Configuring project riscv-gnu-toolchain
    Building project riscv-gnu-toolchain
    ar: `u' modifier ignored since `D' is the default (see `U')
    /tmp/asd-mox/home/mox/uni/s11/Seminar/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/binutils/sysinfo.y: warning: 1 shift/reduce conflict [-Wconflicts-sr]
    syslex_wrap.o: In function `yylex':
    /tmp/asd-mox/home/mox/uni/s11/Seminar/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/build-binutils-newlib/binutils/syslex.c:1097: undefined reference to `yywrap'
    collect2: error: ld returned 1 exit status
    make[3]: *** [Makefile:1372: sysinfo] Error 1
    make[2]: *** [Makefile:3610: all-binutils] Error 2
    make[1]: *** [Makefile:850: all] Error 2
    make: *** [Makefile:200: stamps/build-binutils-newlib] Error 2

aswaterman, 2017-01-13:
Aloha.  This is not a RISC-V-specific issue; rather, a problem with building any binutils on your system.  Take a look at this issue: https://github.com/riscv/riscv-gnu-toolchain/issues/212

closed 2017-01-13 by aswaterman

mox-mox, 2017-01-13:
Thank you for the fast reply. And sorry for not finding that solution myself.
Downgrading flex to 6.6.1-1 worked.