chipsalliance/rocket-chip #516: configuration string pattern mismatch between rocketchip and bbl

sashimi-yzh, 2017-01-18:
[bbl](https://github.com/riscv/riscv-pk/blob/master/machine/configstring.c#L9) uses the pattern `ram{0{addr` to search for memory information. However, the configure string output during elaboration looks like
```
Generated Configuration String
plic {
  priority 0xc000000;
  pending 0xc001000;
  ndevs 2;
};
rtc {
  addr 0x200bff8;
};
core {
  0 {
    0 {
      isa rv64imafds;
      timecmp 0x2004000;
      ipi 0x2000000;
      plic {
        m {
         ie 0xc002000;
         thresh 0xc200000;
         claim 0xc200004;
        };
        s {
         ie 0xc002080;
         thresh 0xc201000;
         claim 0xc201004;
        };
      };
    };
  };
};
ExampleRocketTop {  <= this should be "ram" for bbl to query memory information
  addr 0x80000000;
  size 0x10000000;
}
bootrom {
  addr 0x1000;
  size 0x1000;
}
debug {
  addr 0x0;
  size 0x1000;
}
clint {
  addr 0x2000000;
  size 0x10000;
}
ExampleRocketTop {
  addr 0x60000000;
  size 0x20000000;
}
plic {
  addr 0xc000000;
  size 0x4000000;
}
```

seldridge, 2017-01-18:
FWIW, this is a duplicate of #463 which was superseded by #474. You can hardcode the ram string inside of the `query_mem` function that you reference of `configstring.c` for the time being. 

This is generally bothersome as `pk` is broken out of the box, but it's the state of the world...

aswaterman, 2017-01-18:
Thans @seldridge -- it is bothersome, and @terpstra and I will try to sort out the disagreement on the contents of the config string soon.

sashimi-yzh, 2017-01-19:
Oh, I did not see #474 is going to fix this. Thank you for pointing this out! And it makes sense to do the hardcode. :)

PR #609 (2017-03-25): DTB in the bootrom
When this is merged, it should resolve #516 and #474. The priv-1.10 branch of spike and bbl both support this format.
closed 2017-03-25 by terpstra