chipsalliance/rocket-chip #519: Change AsyncResetReg hack?

mwachs5, 2017-01-18:
We have a hack (due to lack of async reset in Chisel) to blackbox every single AsyncReset register. Previously, we had no Parameterized black boxes, so we had to wrap every single one. Now, @jackkoenig has added (experimental?) parameterized black boxes. This seems like a good use case for it that would save a lot of cruft.

Should I work on using Parameterized black boxes for this? Or is this a waste of time (because experimental, or async is really coming soon, or it will break too many people's constraints)?

mwachs5, 2017-01-18:
@aswaterman @terpstra

aswaterman, 2017-01-18:
I'd guess that proper async reset support is ~3 weeks away.  But if your main goal is to kick the tires on the parameterized black box support, this does seem like a nicely constrained use case.

mwachs5, 2017-09-09:
Closed (for now) by #975

closed 2017-09-09 by mwachs5