chipsalliance/rocket-chip #524: TileLink vs TileLink2

MarcoGeuna, 2017-01-22:
Hi!

I'm studying your repo...and I saw you developed two "similar packages": **_tilelink_** and **_tilelink2_**.
In my knowledge, and reading on internet (unofficial sources!), it seems that tilelink2 is going to replace tilelink. Despite this, I still find it used in important files, like:

- rocket-chip/src/main/scala/rocketchip/**BaseTop.scala**: it uses both tilelink and tilelink2
- freedom/src/main/scala/unleashed/u500vc707devkit/**Top.scala**: it uses only tilelink (sorry, I quote this other repository! But it should not be a problem, since many of you guys work on both!)

Hence: what is the situation about these packages?

Thank you!

hcook, 2017-01-24:
Hello! Thank you for you interest. We are actively converting the codebase from tilelink to the more advanced tilelink2. This conversion will hopefully be entirely complete by the end of February. 

The tilelink2.TLLegacy module provides automatic conversion for masters written against the original TileLink interface, and traces of original tilelink such as bundle definitions will likely remain in the codebase to support e.g. RoCC accelerators that were written against the old standard. In some cases, e.g. BaseTop, the code is already entirely tilelink2-based but we just haven't removed the old import.

MarcoGeuna, 2017-01-24:
Hi @hcook, and thank you for your answer! You clarified the situation for me. But please, let me ask one more question! You said:

> We are also planning to release a new set of documentation for tilelink2, both the protocol and a guide for creating new devices.

Are you planning to release this also by the end of February? Or, can you give me more or less and idea about when this will happen? Thank you!

xfguo, 2017-01-28:
@hcook Waiting for more documentation too. It there any maillist or other methods we can get the news about TileLink2? Thanks

bjoernveit, 2017-02-20:
@hcook I'm currently working on a University Projekt, and struggeling with adding a device, due to the current state, where the tutorial of the README does not work with the current implementation of the project.
Is the realease by the end of the month realistic (for documentation and code)? 
Since we only have limited time to make it work, this is crutial information so please let me know.

edcote, 2017-04-12:
Wondering if there are plans, short term or long term, that will decouple this directory from the rest of rocket-chip?

mwachs5, 2017-09-09:
Tile Link 2 is now documented here: https://www.sifive.com/documentation/tilelink/tilelink-spec/

closed 2017-09-09 by mwachs5