chipsalliance/rocket-chip #539: RoCCExampleConfig Broken, undefined signal in FIRRTL

seldridge, 2017-01-31:
I haven't had a chance to debug it, but the `RoccExampleConfig` isn't getting through FIRRTL:

For completeness, this is using 9c0cc6fdf480efa18a2a4119e6a6e0dfa9235e8e.

```
cd emulator
make CONFIG=RoccExampleConfig

Exception in thread "main" firrtl.passes.PassExceptions: 
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133157.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@133164.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@133165.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133169.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@133176.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@133177.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133181.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Rocc.scala 89:54:rocketchip.RoccExampleConfig.fir@133186.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133225.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@133232.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@133233.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133237.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@133244.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@133245.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133249.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Rocc.scala 89:54:rocketchip.RoccExampleConfig.fir@133254.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133293.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@133300.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@133301.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133305.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@133312.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@133313.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@133317.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Rocc.scala 89:54:rocketchip.RoccExampleConfig.fir@133322.4]: [module LegacyRoccComplex] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@162093.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@162094.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@162098.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@162105.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@162106.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@162110.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@162117.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@162118.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Rocc.scala 51:45:rocketchip.RoccExampleConfig.fir@162121.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@162127.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@162128.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@162132.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@162139.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@162140.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 215:15:rocketchip.RoccExampleConfig.fir@162144.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@162151.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@162152.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Rocc.scala 51:45:rocketchip.RoccExampleConfig.fir@162155.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 216:15:rocketchip.RoccExampleConfig.fir@162161.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
firrtl.passes.CheckChirrtl$UndeclaredReferenceException:  @[Bundles.scala 217:15:rocketchip.RoccExampleConfig.fir@162162.4]: [module RocketTile_l1tol2] Reference _T_177 is not declared.
[... cut ...]
	at firrtl.Compiler$$anonfun$4.apply(Compiler.scala:251)
	at firrtl.Compiler$$anonfun$4.apply(Compiler.scala:250)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.Compiler$class.compile(Compiler.scala:250)
	at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:135)
	at firrtl.Driver$.execute(Driver.scala:143)
	at firrtl.Driver$.execute(Driver.scala:169)
	at firrtl.Driver$.main(Driver.scala:185)
	at firrtl.Driver.main(Driver.scala)
```

colinschmidt, 2017-01-31:
I imagine this config has been broken a while (for my RoCC development I'm pretty far behind master) Perhaps a bisect would help narrow down the issue?

seldridge, 2017-01-31:
No root cause found in rocket-chip source just yet (though this could be Chisel not emitting something it should). Roughly, a wire declaration isn't getting emitted when it should and Chisel isn't catching this, but FIRRTL is. A minimal demonstration of that is with the following (the first instance of the error above):

``` scala
// src/main/scala/uncore/tilelink2/Bundles.scala
object DecoupledSnoop
{
  def apply[T <: Data](source: DecoupledIO[T], sink: DecoupledIO[T]) = {
    val out = Wire(new DecoupledSnoop(sink.bits))
    out.ready := sink.ready // line 215
    out.valid := source.valid
    out.bits  := source.bits
    out
  }
}
```

The resulting FIRRTL with `_T_177` undeclared:
```
    wire _T_457 : {ready : UInt<1>, valid : UInt<1>, bits : {opcode : UInt<3>, param : UInt<3>, size : UInt<4>, source : UInt<3>, address : UInt<32>, mask : UInt<8>, data : UInt<64>}} @[Bundles.scala 214:19]
    _T_457 is invalid @[Bundles.scala 214:19]
    _T_457.ready <= _T_177[0].a.ready @[Bundles.scala 215:15]
    _T_457.valid <= l1tol2_TLHintHandler.io.out[0].a.valid @[Bundles.scala 216:15]
    _T_457.bits <- l1tol2_TLHintHandler.io.out[0].a.bits @[Bundles.scala 217:15]
    _T_360.a <- _T_457 @[Bundles.scala 236:11]
```

@colinschmidt -- working on it! I'm trying to quickly bisect this, but I've started to hit commits on master that don't build. I'll let you know what I find.

Related to this, while bothersome, it's not at all unexpected to find mainline commits that break the build for RoCCs since the regressions aren't running on `RoccExampleConfig`. I think this just further demonstrates the need that at least the example RoCCs need to be built. I'd go further and say that the Proxy Kernel should likely be included in the regression suite.

seldridge, 2017-01-31:
Summary from bisecting:

The first bad commit could be any of:
e22b01a6faa726f7248d7dcecaaea191197dcfca
1b31dfa700be0ed3ce4611212d7a8c0c4e4e15cd
4fe75965a05db2079c999a0c583c29ee1e8d2766
307f938b882f56469ffdaea8bb57e2cdeeb595c9
9a6634cd40085f2721a14950b5fe2ec464553f52
74b6a8d02bf0fce82d5e6ee76069e2208c8e14a5
e0411c6cdec03a0927a4f32adfac401fe663fba0
c1b7c84f090cef1d3c8ccc9c5a935a1b8318d227
We cannot bisect more!

I am not running tests here, just trying to build it and get past the FIRRTL issue outlined above.

The ones that don't build are all something like:
```
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:57)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:526)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:35)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:190)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:190)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:204)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:202)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:202)
	at chisel3.Driver$.elaborate(Driver.scala:190)
Caused by: java.lang.IllegalArgumentException: requirement failed: DefaultCoreplex.TLOutputNode has 1 inputs and 0 outputs; they must match
	at scala.Predef$.require(Predef.scala:219)
	at diplomacy.MixedNode.reqE(Nodes.scala:149)
	at diplomacy.MixedNode.iParams$lzycompute(Nodes.scala:157)
	at diplomacy.MixedNode.iParams(Nodes.scala:155)
	at diplomacy.MixedNode$$anonfun$6.apply(Nodes.scala:156)
	at diplomacy.MixedNode$$anonfun$6.apply(Nodes.scala:156)
	at scala.collection.immutable.List.map(List.scala:273)
	at diplomacy.MixedNode.iParams$lzycompute(Nodes.scala:156)
	at diplomacy.MixedNode.iParams(Nodes.scala:155)
	at diplomacy.MixedNode$$anonfun$6.apply(Nodes.scala:156)
	at diplomacy.MixedNode$$anonfun$6.apply(Nodes.scala:156)
	at scala.collection.immutable.List.map(List.scala:273)
	at diplomacy.MixedNode.iParams$lzycompute(Nodes.scala:156)
	at diplomacy.MixedNode.iParams(Nodes.scala:155)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
```

closed 2017-02-09 by hcook