chipsalliance/rocket-chip #548: 1R1W memories have fantasy behavior and are "doomed to succeed"

proftaylor, 2017-02-06:
https://github.com/ucb-bar/rocket-chip/blob/master/vsim/vlsi_mem_gen:

assign R0_data = ram[reg_R0_addr];  (generated by line 103)

This means that the R0_data output is auto-updated each cycle with the contents of the memory at that address, regardless of whether a valid read was initiated on the previous cycle. For this reason, simulation may work correctly in situations where the actual surrounding logic is incorrect.  This kind of issue is likely to create simulation mismatches when hardened rams are instantiated.

aswaterman, 2017-02-06:
I agree, it would be preferable to return garbage when the read-enable is low and when the same row is written that cycle.  A PR would be welcome.

proftaylor, 2017-02-06:
We tested our own module that assigns 'X to the address register if there
is an address conflict. The hurricane tag definitely has many cases where
simultaneous reads and writes occur to the same location. We are not sure
about the micro-architectural explanation for how this happens and how the
design ensures correct ordering. It looks like in some cases the read is
retried on a subsequent cycle. Perhaps you can shed some light on this?
About half the RISC-V tests appear to stop working in a design based on the
Hurricane tag when we use this replacement module. We are looking into it
but somewhat hampered with a lack of familiarity of the source. It could be
X-pessimism or it could be bugs..

--------

*PASS*:
dhrystone, median, mt-matmul, mt-vvadd, multiply, and vvadd.

*FAIL*:
mm, spmv, qsort, and rsort.

*How it fails?*
mm/spmv --> hang / infinite-loop
qsort/rsort --> exit with errors after verification (whether the sorting
was correct or not)

M

aswaterman, 2017-02-06:
It is likely X-pessimism; try returning $random instead.

aswaterman, 2017-02-06:
I can't speak for all uses of 1R1W RAMs, but I know the ones within Rocket's caches have hazard checks that discard the read value in such cases.

mwachs5, 2017-02-06:
Rocket-chip verilog dies quickly on X... can you assign $random instead?

proftaylor, 2017-02-06:
Will try it!

taylor-bsg, 2017-03-09:
More info. X'ing the outputs of data rams runs a regression successfully. Tags passes the regression only with $random. 

Andrew referred to hazard checks; I am very curious where they might be in the non-blocking data cache? The only one I can find has been commented out:

s1_clk_en := metaReadArb.io.out.valid //TODO: should be metaReadArb.io.out.fire(), but triggers Verilog backend bug

Metadata array is actually single ported and cancels the read if there is a simultaneous write. It sets ready to zero on the read port if there is an incoming write request. However I do not see any use of this signal except in the read arbiter, so I don't see how the pipeline knows that the tag read is bogus... any pointers?

aswaterman, 2017-03-10:
IIRC it's a higher-level microarchitectural property.  Only accesses that own an MSHR can write the tags, and there's at most one MSHR allocated per set.  Any access to a set with an allocated MSHR is either absorbed as a secondary miss or nacked; in neither case does that bogus tag get used.

aswaterman, 2017-03-10:
(It seems probes can write the tags, too, but all accesses are killed when there's a probe in flight.)

terpstra, 2017-03-26:
I am fairly certain that firrtl now uses RANDOMIZE_GARBAGE_ASSIGN to handle this case.

aswaterman, 2017-04-13:
Reopening so I remember to test and incorporate @taylor-bsg's patch:

https://bitbucket.org/taylor-bsg/bsg_riscv/src/689de49a9f0c4a7017f8a138e2f5e973b481db34/patches/rocket-src/nbdcache.scala.patch?at=master&fileviewer=file-view-default

sequencer, 2022-04-08:
any updates on this issue? @aswaterman
I think this is basically the memory read-write contract for SRAM that used in RocketChip which need to be documented in the future.

aswaterman, 2022-04-08:
I don’t think any action is necessary, provided we have improved our RAM models to exhibit pessimistic behavior.