chipsalliance/rocket-chip #549: Multicore rocket failed to run bbl or pk due to unhandlable trap 3

Wonicon, 2017-02-06:
I built `bbl` with an smp-enabled `vmlinux` as payload. It runs well under `spike -p2`. But when I ran it in a multicore rocket chip on zedboard through `./fesvr-zynq bbl`, it failed with such a log:

```
../machine/mtrap.c:15: machine mode: unhandlable trap 3 @ 0x0000000080000270
```

The place where the trap happened is in `riscv-tools/riscv-pk/machine/mentry.S`, right after the `wfi` instruction:

```
.LmultiHart:
#if MAX_HARTS > 1
  # wait for an IPI to signal that it's safe to boot
  wfi 
  csrr a1, mip 
  andi a1, a1, MIP_MSIP
  beqz a1, .LmultiHart

  # make sure our hart id is within a valid range
  fence
  li a1, MAX_HARTS
  bltu a0, a1, init_other_hart
#endif
```

I find the `bad_trap()` causing the failure is called from `trap_from_machine_mode()` in `riscv-tools/riscv-pk/machine/mtrap.c`. If I let this function directly return, the riscv logo can be shown, then the system gets stuck. Therefore, I think the trap is received by the second core, and the repeated triggering of trap 3 blocks the second core.

I find trap 3 corresponds to breakpoint, but I have no idea what triggers this interruption in machine mode. I tried to configure rocket chip with `UseDebug => false`, but the problem still exists.

I am using `fpga-zynq` repo ([commit ac39be1](https://github.com/ucb-bar/fpga-zynq/commit/ac39be1c1f85df271becf1cd410eab49914ca940)), the `rocket-chip` is `fpga-zynq`'s corresponding submodule version. Linux is version 4.6.2.

Originally I used the `riscv-tools` in master branch ([commit cd78e37](https://github.com/riscv/riscv-tools/commit/cd78e37f72cfc2a452a0c11744586084fbae1dcd)). Now I am trying `riscv-tools` repo that `fpga-zynq`'s `rocket-chip` depends on as a submodule, but problem still exists. The `fesvr-zynq` has alos been rebuilt.

To obtain multicore environment on zedboard, I modify the `WithSmallCores` as follows:

```
class WithSmallCores extends Config(
  (pname, site, here) => pname match {
    case NTLBEntries => 4
    case BtbKey => BtbParameters(nEntries = 0)
    case NTiles => {
        2
    }   
    case _ => throw new CDEMatchError
  },  
  knobValues = { 
    case "L1D_SETS" => 64
    case "L1D_WAYS" => 1
    case "L1I_SETS" => 64
    case "L1I_WAYS" => 1
    case "L1D_MSHRS" => 0
    case _ => throw new CDEMatchError
  })
```

Wonicon, 2017-02-06:
I find that when exec `hls->ipi = 1` in `riscv-toos/riscv-pk/machine/configstring.c` for the second core's hart
will cause that core receive interruption 3 (breakpoint).

My config for hart is:

```
hart0: 0x0000000002000000
hart1: 0x0000000002000004
```

terpstra, 2017-03-26:
I am pretty sure from your description that this is fallout from the move from config string to FDT. For quite some time, config string in rocketchip was broken, but bbl required it. At this point the priv-1.10 branch of bbl should work with the output of rocketchip. I am going to confirm this with our own FPGA builds this next week, but if you could confirm this resolves the issue for your zedboard, I'd appreciate it.

terpstra, 2017-05-17:
This has been working reliably on our boards for the past months. Since you haven't responded, I am going to assume that when you switched to using DTB in both hardware and software, the problem went away, as it's pretty clear configstring.c was not going to work with current rocketchip

closed 2017-05-17 by terpstra

Wonicon, 2017-05-19:
I'm sorry for my no response. My working environment has changed and it is not convenient to test on zedboard any more. We isolate the tiles' memory space and let them run standalone kernels as a workaround. But SMP is still a need in our system. So I really appreciate your work.