chipsalliance/rocket-chip #553: Skip Travis builds on non-functional PRs

jackkoenig, 2017-02-09:
PRs like https://github.com/ucb-bar/rocket-chip/pull/530 just eat Travis bandwidth.

This could probably be accomplished by adding a tag to commits like [non-functional] and having the Travis script just pass if it sees that. This of course relies on reviewers verifying that the changes are non-functional.

jackkoenig, 2017-02-14:
It looks like this functionality exists by putting `[ci skip]` or `[skip ci]`: https://docs.travis-ci.com/user/customizing-the-build#Skipping-a-build

In my opinion, we should take advantage of this.

terpstra, 2017-03-26:
We have gotten travis build times back down to 35 minutes and we only build on PR pushes, so I think wasted Travis bandwidth is not much of an issue anymore. Nevertheless, yes we should endeavor not to waste build time either by marking commits as you suggested or just not pushing a branch to a PR unless we want it built (which is what I think most people are doing).

closed 2017-03-26 by terpstra