chipsalliance/rocket-chip #571: Hasti can't  handle N masters to one slave

colin4124, 2017-03-07:
First of all, let's look at one piece of code in [hasti.scala](https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/junctions/hasti.scala#L280):

``` scala
  for (m <- 0 until nMasters) {
    // If the master is connected to a slave, the slave determines hready.
    // However, if no slave is connected, for progress report ready anyway, if:
    //   bad address (swallow request) OR idle (permit stupid masters to move FSM)
    val autoready = nowhereM(m) || masters(m).isIdle()
    val hready = Mux1H(unionGrantMS(m), slaves.map(_.hready ^ autoready)) ^ autoready
    masters(m).hready := hready
    // If we diverted a master, we need to absorb his address phase to replay later
    diversions(m).io.divert := (bubbleM(m) || blockedM(m)) && NSeq(m) && hready
  }
```

When the number of masters is N and slaves is only one,  `unionGrantMS(m)` has only one element, the same as `slaves.map(_.hready ^ autoready)` . As the result, two parameters of `Mux1H` both are the vector that contains only one element.

Then let's look at the implement of `Mux1H` in [Mux.scala](https://github.com/ucb-bar/chisel3/blob/master/src/main/scala/chisel3/util/Mux.scala#L16):

``` scala
/** Builds a Mux tree out of the input signal vector using a one hot encoded
  * select signal. Returns the output of the Mux tree.
  *
  * @note results undefined if multiple select signals are simultaneously high
  */
object Mux1H {
  def apply[T <: Data](sel: Seq[Bool], in: Seq[T]): T =
    apply(sel zip in)
  def apply[T <: Data](in: Iterable[(Bool, T)]): T = SeqUtils.oneHotMux(in)
  def apply[T <: Data](sel: UInt, in: Seq[T]): T =
    apply((0 until in.size).map(sel(_)), in)
  def apply(sel: UInt, in: UInt): Bool = (sel & in).orR
}
```

Mux1H will pass his two parameters as a vector of tuple, witch contains only one tuple, to [SeqUtils.oneHotMux](https://github.com/ucb-bar/chisel3/blob/master/chiselFrontend/src/main/scala/chisel3/core/SeqUtils.scala#L54) : 

``` scala
  /** Returns the data value corresponding to the lone true predicate.
    *
    * @note assumes exactly one true predicate, results undefined otherwise
    */
  def oneHotMux[T <: Data](in: Iterable[(Bool, T)]): T = macro CompileOptionsTransform.inArg

  def do_oneHotMux[T <: Data](in: Iterable[(Bool, T)])(implicit sourceInfo: SourceInfo, compileOptions: CompileOptions): T = {
    if (in.tail.isEmpty) {
      in.head._2
    } else {
      val masked = for ((s, i) <- in) yield Mux(s, i.asUInt, 0.U)
      val output = cloneSupertype(in.toSeq map {_._2}, "oneHotMux")
      output.fromBits(masked.reduceLeft(_|_))
    }
  }
```

Because `in` contains only one element, `in.tail.isEmpty` is true and then it returns the second element of the tuple. **In other words, if input signal vector contains only one element, Mux1H will always see it as true and return the second parameter.**

Take back to `val hready = Mux1H(unionGrantMS(m), slaves.map(_.hready ^ autoready)) ^ autoready`, **all the masters will ready at the same time if the only one slave ready.**
 
Is this a bug of Mux1H? If not, what the situation is when` if (in.tail.isEmpty) { in.head._2 }`

colin4124, 2017-03-09:
I have fixed this by pulling request in chisel3 https://github.com/ucb-bar/chisel3/pull/546  

But I can't confirm this, if chisel3 accepts my PR, I will close this issue.

aswaterman, 2017-03-09:
This should be fixed here in rocket-chip. oneHotMux's behavior is correct as-is.

closed 2017-03-14 by hcook

colin4124, 2017-03-16:
We should take `Mux1H` seriously by considering its boundary condition. If input signal vector contains only one element, `Mux1H` will always see it as true and return the second parameter. There is another  careless usage of `Mux1H` in [hasti.scala](https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/junctions/hasti.scala#L295) as following.
```
// Master muxes (address and data phase are the same)
(masters zip unionGrantMS) foreach { case (m, g) => {
    m.hrdata := Mux1H(g, slaves.map(_.hrdata))
    m.hresp  := Mux1H(g, slaves.map(_.hresp))
} }
```
When there is one slave, all masters will receive the slave's response whether requested the slave  or not. Thankfully, master would ignore that if **m.hready**  is not ready.
```
// Slave address phase muxes
  (slaves zip addressPhaseGrantSM) foreach { case (s, g) => {
    s.htrans    := Mux1H(g, masters.map(_.htrans))
    s.haddr     := Mux1H(g, masters.map(_.haddr))
    s.hmastlock := isLocked
    s.hwrite    := Mux1H(g, masters.map(_.hwrite))
    s.hsize     := Mux1H(g, masters.map(_.hsize))
    s.hburst    := Mux1H(g, masters.map(_.hburst))
    s.hprot     := Mux1H(g, masters.map(_.hprot))
    s.hsel      := g.reduce(_ || _)
} }
```
When there is one master, all slaves will receive the master's request whether the master requested the slave  or not. Thankfully, slave would ignore that if **s.hsel**  is clear.

Thus, should I fix these **warnings** as https://github.com/ucb-bar/rocket-chip/pull/576 ?

@aswaterman @terpstra

terpstra, 2017-03-16:
I don't think these uses of Mux1H are wrong. The cases where it was used were intentionally situations where it is safe to connect it directly. However, this code is obsolete anyway.

colin4124, 2017-03-16:
@terpstra I got it. Thinks very much. ^_^