chipsalliance/rocket-chip #577: build error while generate verilog for FPGA config.

paladinw, 2017-03-09:
build error while generate verilog for FPGA config.

paladinw@ubuntu:~/mine/riscv/rocket-chip/vsim$ make verilog CONFIG=DefaultFPGAConfig
mkdir -p /home/paladinw/mine/riscv/rocket-chip/vsim/generated-src/
cd /home/paladinw/mine/riscv/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/paladinw/mine/riscv/rocket-chip/sbt-launch.jar "run-main rocketchip.Generator /home/paladinw/mine/riscv/rocket-chip/vsim/generated-src rocketchip TestHarness rocketchip DefaultFPGAConfig"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /home/paladinw/mine/riscv/rocket-chip/project
Using addons: 
[info] Set current project to rocketchip (in build file:/home/paladinw/mine/riscv/rocket-chip/)
[info] Compiling 1 Scala source to /home/paladinw/mine/riscv/rocket-chip/chisel3/target/scala-2.11/classes...
[info] Packaging /home/paladinw/mine/riscv/rocket-chip/chisel3/target/scala-2.11/chisel3_2.11-3.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Running rocketchip.Generator /home/paladinw/mine/riscv/rocket-chip/vsim/generated-src rocketchip TestHarness rocketchip DefaultFPGAConfig
[info] [0.004] Elaborating design...
[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:42)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:236)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:234)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:234)
	at chisel3.Driver$.elaborate(Driver.scala:91)
Caused by: java.lang.IllegalArgumentException: requirement failed
	at scala.Predef$.require(Predef.scala:207)
	at uncore.devices.DebugModuleConfig.<init>(Debug.scala:221)
	at uncore.devices.DefaultDebugModuleConfig.<init>(Debug.scala:245)
	at coreplex.BaseCoreplexConfig$$anonfun$$lessinit$greater$1$$anonfun$apply$1.applyOrElse(Configs.scala:25)
	at scala.runtime.AbstractPartialFunction.apply(AbstractPartialFunction.scala:36)
	at config.PartialParameters.chain(Config.scala:59)
	at config.Config.chain(Config.scala:32)
	at config.ChainParameters.chain(Config.scala:49)
	at config.Config.chain(Config.scala:32)
	at config.ChainView.find(Config.scala:45)
	at config.EmptyParameters.chain(Config.scala:53)
	at config.Config.chain(Config.scala:32)
	at config.ChainParameters.chain(Config.scala:49)
	at config.Config.chain(Config.scala:32)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1
[error] Total time: 8 s, completed Mar 9, 2017 7:18:41 AM
/home/paladinw/mine/riscv/rocket-chip/vsim/Makefrag-verilog:12: recipe for target '/home/paladinw/mine/riscv/rocket-chip/vsim/generated-src/rocketchip.DefaultFPGAConfig.fir' failed
make: *** [/home/paladinw/mine/riscv/rocket-chip/vsim/generated-src/rocketchip.DefaultFPGAConfig.fir] Error 1

paladinw, 2017-03-09:
no issues while run $> make verilog

terpstra, 2017-03-26:
The problem appears to be that FPGAConfig does not include any processors, and the Debug unit expects to be connected to at least one processor.

@hcook What is the intended behaviour of FPGAConfig? Should it include a processor? I note that DefaultFPGASmallConfig includes a 'SmallCore'. Perhaps we should add a 'BigCore' to the default?

terpstra, 2017-04-15:
@paladinw just use: make verilog CONFIG=DefaultFPGASmallConfig

closed 2017-05-17 by terpstra