chipsalliance/rocket-chip #579: Unable to  "make run- asm -tests" in emulator

nobueno, 2017-03-10:
Hello,
I have made the RISCV tools. Added the RISCV and riscvtools/bin to the PATH. Now, when I try to run the asm tests, I get error. 
`{txace3:emulator} make
mkdir -p /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig
/proj/txace/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/proj/txace/rocket-chip/vsrc -O0 -CFLAGS "-O1 -std=c++0x -I/proj/txace/RISC-V/include -I/proj/txace/rocket-chip/csrc -DVERILATOR -include /proj/txace/rocket-chip/csrc/verilator.h" -Mdir /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig \
	-o /proj/txace/rocket-chip/emulator/emulator-rocketchip-DefaultConfig /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /proj/txace/rocket-chip/csrc/emulator.cc /proj/txace/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/proj/txace/RISC-V/lib -Wl,-rpath,/proj/txace/RISC-V/lib -L/proj/txace/rocket-chip/emulator -lfesvr -lpthread" \
	-CFLAGS "-I/proj/txace/rocket-chip/emulator/generated-src -include /proj/txace//rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h -include /proj/txace/rocket-chip/emulator/generated-src/consts.DefaultConfig.h"
make VM_PARALLEL_BUILDS=1 -C /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig -f VTestHarness.mk
make[1]: Entering directory `/proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig'
g++    SimDTM.o emulator.o verilated.o verilated_dpi.o VTestHarness__ALL.a   -L/proj/txace/RISC-V/lib -Wl,-rpath,/proj/txace/RISC-V/lib -L/proj/txace/rocket-chip/emulator -lfesvr -lpthread  -o /proj/txace/rocket-chip/emulator/emulator-rocketchip-DefaultConfig -lm -lstdc++  2>&1 | c++filt
/proj/txace/RISC-V/lib/libfesvr.so: undefined reference to `std::__throw_bad_function_call()@GLIBCXX_3.4.14'
collect2: ld returned 1 exit status
make[1]: Leaving directory `/proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig'
`

This seems like a compiler issue, but I don't see any solution posted.

paladinw, 2017-03-10:
I can build and run the test with latest code.
My OS is ubuntu16.04

nobueno, 2017-03-10:
Mine (the server I am running on) is CentOS
```
{txace3:rocket-chip} lsb_release -a
LSB Version:	:base-4.0-amd64:base-4.0-noarch:core-4.0-amd64:core-4.0-noarch:graphics-4.0-amd64:graphics-4.0-noarch:printing-4.0-amd64:printing-4.0-noarch
Distributor ID:	CentOS
Description:	CentOS release 6.8 (Final)
Release:	6.8
```

Also, if you notice, it shows the error with the libfesvr.so file which was generated by the riscv-tools. I am able to "make verilog" (under the vsim folder), but also in there if I try to run the tests, it fails.

nobueno, 2017-03-16:
Hello All, 
Can someone help me with issues with the rocket-chip build. I am still using the same OS as mentioned in the previous message. I re-downloaded the rocket-chip and rebuilt the riscv-tools using GCC4.8.
The helloworld test runs fine. But then I went to "emulator" and trying to "make" it. After elaboration, I get this error:
`[info] [21.730] Done elaborating.
[success] Total time: 142 s, completed Mar 16, 2017 12:04:31 PM
mkdir -p /proj/txace/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /proj/txace/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.fir -o /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v -X verilog
mkdir -p /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig
/proj/txace/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/proj/txace/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/proj/txace/rocket-chip/riscv_toolchain/include -I/proj/txace/rocket-chip/csrc -DVERILATOR -include /proj/txace/rocket-chip/csrc/verilator.h" -Mdir /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig \
	-o /proj/txace/rocket-chip/emulator/emulator-rocketchip-DefaultConfig /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /proj/txace/rocket-chip/csrc/emulator.cc /proj/txace/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/proj/txace/rocket-chip/riscv_toolchain/lib -Wl,-rpath,/proj/txace/rocket-chip/riscv_toolchain/lib -L/proj/txace/rocket-chip/emulator -lfesvr -lpthread" \
	-CFLAGS "-I/proj/txace/rocket-chip/emulator/generated-src -include /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h"
/proj/txace/rocket-chip/emulator/verilator/install/bin/verilator_bin: /proj/cad/cadence/soc-8.10.004.lnx86/tools/lib/64bit/libstdc++.so.6: version `GLIBCXX_3.4.11' not found (required by /proj/txace/rocket-chip/emulator/verilator/install/bin/verilator_bin)
/proj/txace/rocket-chip/emulator/verilator/install/bin/verilator_bin: /proj/cad/cadence/soc-8.10.004.lnx86/tools/lib/64bit/libstdc++.so.6: version `GLIBCXX_3.4.9' not found (required by /proj/txace/rocket-chip/emulator/verilator/install/bin/verilator_bin)
%Error: Command Failed /proj/txace/rocket-chip/emulator/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/proj/txace/rocket-chip/vsrc -O3 -CFLAGS '-O1 -std=c++11 -I/proj/txace/rocket-chip/riscv_toolchain/include -I/proj/txace/rocket-chip/csrc -DVERILATOR -include /proj/txace/rocket-chip/csrc/verilator.h' -Mdir /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig -o /proj/txace/rocket-chip/emulator/emulator-rocketchip-DefaultConfig /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /proj/txace/rocket-chip/csrc/emulator.cc /proj/txace/rocket-chip/csrc/SimDTM.cc -LDFLAGS ' -L/proj/txace/rocket-chip/riscv_toolchain/lib -Wl,-rpath,/proj/txace/rocket-chip/riscv_toolchain/lib -L/proj/txace/rocket-chip/emulator -lfesvr -lpthread' -CFLAGS '-I/proj/txace/rocket-chip/emulator/generated-src -include /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h'
make: *** [emulator-rocketchip-DefaultConfig] Error 1
`
After changing the LD_LIBRARY_PATH to gcc4.8.2/lib64, I get a new error:
`[info] [7.627] Done elaborating.
[success] Total time: 17 s, completed Mar 16, 2017 12:19:10 PM
mkdir -p /proj/txace/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /proj/txace/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.fir -o /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v -X verilog
mkdir -p /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig
/proj/txace/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/proj/txace/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/proj/txace/rocket-chip/riscv_toolchain/include -I/proj/txace/rocket-chip/csrc -DVERILATOR -include /proj/txace/rocket-chip/csrc/verilator.h" -Mdir /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig \
	-o /proj/txace/rocket-chip/emulator/emulator-rocketchip-DefaultConfig /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /proj/txace/rocket-chip/csrc/emulator.cc /proj/txace/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/proj/txace/rocket-chip/riscv_toolchain/lib -Wl,-rpath,/proj/txace/rocket-chip/riscv_toolchain/lib -L/proj/txace/rocket-chip/emulator -lfesvr -lpthread" \
	-CFLAGS "-I/proj/txace/rocket-chip/emulator/generated-src -include /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h"
make VM_PARALLEL_BUILDS=1 -C /proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig -f VTestHarness.mk
[... cut ...]
make[1]: *** [SimDTM.o] Error 1
cc1plus: error: unrecognized command line option "-std=c++11"
cc1plus: warning: unrecognized command line option "-Wno-parentheses-equality"
cc1plus: warning: unrecognized command line option "-Wno-parentheses-equality"
make[1]: *** [verilated.o] Error 1
make[1]: *** [verilated_dpi.o] Error 1
make[1]: Leaving directory `/proj/txace/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig'
make: *** [emulator-rocketchip-DefaultConfig] Error 2
`

Do i need "root" privilege for this?

aswaterman, 2017-03-16:
The ```-std=c++11``` error suggests you aren't actually using g++ 4.8 here.

nobueno, 2017-03-16:
How do I use the g++4.8 here? I couldn’t find the Makefile where the CC or CXX is defined. I noticed it calls the “Makefrag” from the base rocket-chip directory where there is a declaration for CXX. I did make the change there before running the “make” in emulator, still same results..

   

      1 # check RISCV environment variable

      2 ifndef RISCV

      3 $(error Please set environment variable RISCV. Please take a look at README)

      4 endif

      5 

      6 MODEL ?= TestHarness

      7 PROJECT ?= rocketchip

      8 CFG_PROJECT ?= $(PROJECT)

      9 CONFIG ?= DefaultConfig

     10 # TODO: For now must match rocketchip.Generator

     11 long_name = $(PROJECT).$(CONFIG)

     12 

     13 CXX = g++4.8   #Monir Change: Original line was CXX ?= g++

     14 CXXFLAGS := -O1

     15 

 

For the record, the riscv-tools I built, I manually defined the gcc4.8 and g++4.8 by the following command:

 

$ CC=gcc4.8 C++=g++4.8 ./build.sh

 

 

I get the same –std=c++11 for “vsim” as well.

 

Thanks in advance.

aswaterman, 2017-03-16:
Verilator invokes gcc; you could check their documentation to see if there's a straightforward way.  One option is to create a symlink from g++ to g++-4.8 and put that symlink in your PATH.

nobueno, 2017-03-16:
I thought about the symlink option. But currently “gcc” is defaulted at /usr/bin/gcc. If I have to symlink, I need to tak the gcc4.8 and create a symlink of that file in /usr/bin/gcc, which creates conflict. I can’t remove the existing /usr/bin/gcc cause I don’t have root permission on the server. 

 

Same thing is happening for VCS. Both Verilator and VCS is calling for gcc/g++ which are in /usr/bin/ location.

 

 

/Monir

aswaterman, 2017-03-16:
You could put it in a local directory, and put that directory at the front of your PATH.  It doesn't have to go in /usr/bin.

nobueno, 2017-03-16:
Thanks Andrew. 
Got that working (i think) after symlinking. But now have error related to GLIBCXX_3.4.14 when i run "make run":

`ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/spmv.riscv output/spmv.riscv
mkdir -p ./output
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/mt-vvadd.riscv output/mt-vvadd.riscv
mkdir -p ./output
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/median.riscv output/median.riscv
mkdir -p ./output
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/multiply.riscv output/multiply.riscv
mkdir -p ./output
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/qsort.riscv output/qsort.riscv
mkdir -p ./output
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/towers.riscv output/towers.riscv
mkdir -p ./output
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/vvadd.riscv output/vvadd.riscv
mkdir -p ./output
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/dhrystone.riscv output/dhrystone.riscv
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64uf-p-ldst 3>&1 1>&2 2>&3 | /proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm  > output/rv64uf-p-ldst.out && [ $PIPESTATUS -eq 0 ]
ln -fs /proj/txace/rocket-chip/riscv_toolchain/riscv64-unknown-elf/share/riscv-tests/benchmarks/mt-matmul.riscv output/mt-matmul.riscv
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64uf-p-move 3>&1 1>&2 2>&3 | /proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm  > output/rv64uf-p-move.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64uf-p-fsgnj 3>&1 1>&2 2>&3 | /proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm  > output/rv64uf-p-fsgnj.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64uf-p-fcmp 3>&1 1>&2 2>&3 | /proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm  > output/rv64uf-p-fcmp.out && [ $PIPESTATUS -eq 0 ]
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libriscv.so)
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libfesvr.so)
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libriscv.so)
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libfesvr.so)
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libriscv.so)
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libfesvr.so)
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libriscv.so)
/proj/txace/rocket-chip/riscv_toolchain/bin/spike-dasm: /usr/lib64/libstdc++.so.6: version `GLIBCXX_3.4.14' not found (required by /proj/txace/rocket-chip/riscv_toolchain/lib/libfesvr.so)
make: *** [output/rv64uf-p-ldst.out] Error 1
make: *** Waiting for unfinished jobs....
make: *** [output/rv64uf-p-move.out] Error 1
make: *** [output/rv64uf-p-fsgnj.out] Error 1
make: *** [output/rv64uf-p-fcmp.out] Error 1
{txace3:emulator} echo $LD_LIBRARY_PATH
/proj/txace/rocket-chip/riscv_toolchain/lib64
`

nobueno, 2017-03-16:
never mind, after changing the LD_LIBRARY_PATH, i think it is working.

closed 2017-03-24 by hcook