chipsalliance/rocket-chip #585: How to enable interrupt? How to execute from external cacheable space?

LouisGaoNXP, 2017-03-14:
Here is my Generated Address Map:
        plic c000000 - 10000000, RW
        debug 0 - 1000, RWX
        zeros_0 a000000 - c000000, RWX [C]
        ExampleRocketTop 20000000 - 40000000, RWX
        clint 2000000 - 2010000, RW
        bootrom 1000 - 2000, RX
        ExampleRocketTop 80000000 - c0000000, RWX [C]

I have two issues:
1. How to enable interrupt?
    I run simulation in my testbench using generated verilog design file rocketchip.DefaultConfig.v. I forced the ExampleRocketTop's input io_interrupts_0_0 to high, but Core cannot enter my interrupt routine. The following is my .c code.

   unsigned long main( void )
{

    //Set trap vector table
       __asm("li a1, 0x2EFC1000");
       __asm("CSRRW   x0, mtvec, a1");

    //Enable interrupt
       __asm("li a1, 0xF0F");
       __asm("csrw mie, a1");

    //Software pending 
    //   __asm("li a1, 0x008");
    //   __asm("csrw mip, a1");

    while (1);
    return 0xdead0001;                    /* done, test fails, should've gone to isr */
}

2. How to execute from the cachable space 80000000 - c0000000?
    I put my main() code in the space from 0x9000_0000, and let ROM code jump to 0x9000_0000. But after the jumping, Rocket core seems not execute my main() code. 
When I put my main() code in the space  20000000 - 40000000, everything works well.

Is there any document about the memory map of RocketChip, and registers of PLIC?

(I'm new to Scala lauguage. I tried to understand source code, but it's so difficult for me...)

apolonic, 2017-08-16:
@LouisGaoNXP  Sir, do you have more information on the interrupt issue?

sequencer, 2022-04-08:
I thought this should refer to PLIC/CLIC(CLINT) spec.
But essentially, we need a test case (depends on #2953) to demonstrate this.

michael-etzkorn, 2022-06-04:
A test case would be nice. 
It'd also be good to have an example design flow that includes setting this up with the PLIC regmap. 

Given this is a StackOverflow style question, I'll back-link #2960 where we can add a TODO for a tutorial example flow and close this. 

If either apolonic and Louis or anyone viewing this issue wants to revisit this, I'd encourage posing questions to Stack Overflow about what they've tried, and I'd be willing to throw some guidance there.

closed 2022-06-04 by michael-etzkorn