chipsalliance/rocket-chip #596: How to build the RoCC sample examples

nitish2112, 2017-03-20:
Apparently building the sample RoCC examples using:

% cd vsim
% make make CONFIG=WithRoccExample

do not produce the RTL for RoCC. Is there a patch that needs to be applied? What is the best commit point for using sample RoCC examples?

seldridge, 2017-03-20:
See #558. #559 gets you to (or it did) support for one RoCC, but can't handle multi-RoCC builds.

While purely anecdotal, I'm sitting on a merge-base of c531093898f1f4ff22d1c2cf9d5bb2310d05fc31 with a slightly newer version of riscv-tools (https://github.com/riscv/riscv-tools/tree/cd78e37f72cfc2a452a0c11744586084fbae1dcd). This is roughly pre-Legacy RoCC modifications. I'm just sitting here until #558 gets fixed or I have time to get #559 to work for multi-RoCCs.

nitish2112, 2017-03-21:
I am at commit point d4272db and made the changes in src/main/scala/coreplex/Configs.scala by replacing the class WithRoCCExample with the one in #558 but while compiling to verilog I am getting this error:

`[warn]   debug(idx)
[warn]   ^
[warn] there were 3180 feature warnings; re-run with -feature for details
[warn] two warnings found
[info] Running rocketchip.Generator /ufs/vlsi/nks45/RoCC/rocket-chip/vsim/generated-src rocketchip TestHarness rocketchip WithRoccExample
**[error] (run-main-0) java.lang.Exception: Unable to find part "rocketchip.WithRoccExample" from "WithRoccExample", did you misspell it?**
java.lang.Exception: Unable to find part "rocketchip.WithRoccExample" from "WithRoccExample", did you misspell it?
	at Chisel.package$throwException$.apply(compatibility.scala:208)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:32)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:27)
	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
	at util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:27)
	at rocketchip.Generator$.getConfig(Generator.scala:11)
	at util.GeneratorApp$class.config(GeneratorUtils.scala:79)
	at rocketchip.Generator$.config$lzycompute(Generator.scala:11)
	at rocketchip.Generator$.config(Generator.scala:11)
	at util.GeneratorApp$class.world(GeneratorUtils.scala:80)
	at rocketchip.Generator$.world$lzycompute(Generator.scala:11)
	at rocketchip.Generator$.world(Generator.scala:11)
Caused by: java.lang.ClassNotFoundException: rocketchip.WithRoccExample
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:264)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:29)
	at util.HasGeneratorUtilities$$anonfun$getConfig$1.apply(GeneratorUtils.scala:27)
	at scala.collection.IndexedSeqOptimized$class.foldr(IndexedSeqOptimized.scala:62)
	at scala.collection.IndexedSeqOptimized$class.foldRight(IndexedSeqOptimized.scala:70)
	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:48)
	at util.HasGeneratorUtilities$class.getConfig(GeneratorUtils.scala:27)
	at rocketchip.Generator$.getConfig(Generator.scala:11)
	at util.GeneratorApp$class.config(GeneratorUtils.scala:79)
	at rocketchip.Generator$.config$lzycompute(Generator.scala:11)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1
[error] Total time: 173 s, completed Mar 20, 2017 8:50:25 PM
`

closed 2017-07-21 by colinschmidt