chipsalliance/rocket-chip #601: AsyncQueue Question

shunshou, 2017-03-20:
I'm trying to use the rocket-chip util AsyncQueue for a project, and I'm running into some trouble.

Scenario is: 

enq_clk at some clk frequency
deq_clk either at the same clk frequency (shifted in phase) or at a slightly higher clk frequency (less than double)

While enq_valid is low, I pulse enq_reset, deq_reset for one deq_clk period (which should be asynchronous?)

Then, at the earliest, on the next, next enq_clk after the resets are deasserted, I start feeding in valid data to the queue (and keep valid asserted).

I've also made deq_ready to be always true (since my receiving end is always at the faster clk rate). 

---

Since my receiver is always at the faster clk rate, theoretically, I think I only need a depth of 2? But for whatever reason, it looks like (?) the queue fills up and then deq_valid goes low after I count depth # of deq_valid's. 

---

Also, what does "safe = true" do? I see it dramatically affects the latency. Also, with "safe = true", it looks like the queue is missing the first N (where N is kind of large) valid enq_data. 

Even with "safe = false", some of my queue's seem to be missing the first valid data following the reset. 

Am I missing something? Any help would be super appreciated.

mwachs5, 2017-03-20:
You should also be only feeding in data if enq_ready is true... are you checking that? Basically the queue takes a while to come out of reset.

shunshou, 2017-03-20:
Ok, I wasn't checking enq_ready. Besides right after reset, is there anything that would cause enq_ready to go down? Like after it resets, it should always be high, I presume? Otherwise, this wouldn't work for a constantly streaming input.

mwachs5, 2017-03-20:
That seems correct to me.

mwachs5, 2017-03-20:
Actually no that isn't right... the queue has some synchronizer depth that you need to overcome. Like it may take several cycles for the other side to realize that you have put data in and then communicate back that it has taken out.

In other words the queue is high-latency. You need enough depth to overcome that. If you have enough depth, then what you say is true.

shunshou, 2017-03-20:
1) What is "enough depth"?
2) So with enough depth, you're implying that I can then handle constant input stream and not miss data?

aswaterman, 2017-03-20:
I believe the depth required to maintain full throughput is 2*(sync+1), but I'm not 100% sure about that.  In any case, past whatever that threshold is, a consumer whose clock is at least as fast as the producer will never lose data.

shunshou, 2017-03-20:
the 2*(sync+1) would make sense wrt the default settings on AsyncQueue. And everything looks a lot better now. Thanks so much for the help! :)

shunshou, 2017-03-21:
FYI seems like 8 wasn't sufficient (enq_valid was periodically going low), but 16 did get a frame of ~800 streamed inputs w/o missing data. 

I realize this code prob isn't meant to be used outside of rocket-chip, but it'd probably be useful to state that somewhere.

Anyways, thanks for the help! :)

closed 2017-03-22 by hcook