chipsalliance/rocket-chip #644: rocket-chip/riscv-tools spike pk hello : fail to run , no such file or directory , child dtb process failed

Peach92, 2017-04-02:
My os is Ubuntu 14.04 gcc5.4 g++5.4 

git clone https://github.com/ucb-bar/rocket-chip.git
cd rocket-chip
git submodule update --init --recursive
cd riscv-tools
export RISCV=/rocket-chip folder/riscv-tools/riscv
./build.sh

It says something like '.....Toolchain compledted....'
I can compile hello.c by using the command "riscv64-unknown-elf-gcc -o hello hello.c" The hello object file is sucessfully generated.
However, when I execute "spike pk hello" it returns "Failed to run : No such file or directory Child dtb process failed"

I tried at least thress times and have the same problem every time.

aswaterman, 2017-04-02:
You need to install dtc.  On Ubuntu, use ```sudo apt-get install device-tree-compiler```.

closed 2017-04-02 by aswaterman

Peach92, 2017-04-03:
Thanks. I am very grateful for your help. but when I 
sudo apt-get install device-tree-compiler
./build.sh

It returns another "Failed to run : No such file or directory Child dts process failed"