chipsalliance/rocket-chip #649: spike pk hello: terminate called after throwing an instance of 'trap_store_access_fault'

tinna0529, 2017-04-04:
Ubuntu 16.04.4
gcc 5.4

./build.sh
RISC-V Toolchain installation completed!

spike pk hello
terminate called after throwing an instance of 'trap_store_access_fault'

hauhsu, 2017-04-11:
You are not providing enough info.
Did you instal the current RISC-V Toolchain?
Are you sure you are running spike with pk?
You should follow the instructions in the following link to compile C program and run spike:
https://github.com/riscv/riscv-isa-sim#compiling-and-running-a-simple-c-program

Also, I think this issue should be opened in the [spike repo](https://github.com/riscv/riscv-isa-sim) instead of here.

tinna0529, 2017-04-12:
I change the VM OS from ubuntu-16.04-desktop-i386 to ubuntu-14.04.5-desktop-amd64.
  gcc g++ 4.8.4
  apt-get install device-tree-compiler
  
  so the spike is OK!

closed 2017-04-12 by tinna0529