chipsalliance/rocket-chip #652: Javac not found while making the emulator

ninode, 2017-04-05:
Hello, 

While running make in the emulator directory, I've got an error: Cannot run javac in //rocket-chip/firrtl no such file or directory. Could you tell me what this is needed for ? 

Thank you

closed 2017-04-05 by ninode

olix86, 2018-01-26:
Hi @ninode I'm stuck with the same problem, did you manage to resolve it?

Thanks!

ninode, 2018-01-27:
I think you could try to check if jdk is installed, run: javac.

If jdk isn't installed, then try: sudo apt-get install openjdk-7-jdk.
Then check if javac is in your PATH, if it isn't, add export PATH=${PATH}:/usr/lib/jvm/java-7-open-jdk/bin
Next, sudo ln -s /usr/lib/jvm/java-7-open-jdk/bin/javac /usr/bin/ 
Next, PATH=${PATH}:/usr/lib/jvm/java-7-open-jdk/bin/javac

Hopefully this works !