chipsalliance/rocket-chip #653: L2 Cache Missing?

sxu55, 2017-04-05:
Has L2 cache been removed from the rocket chip generator recently? Originally I saw L2 cache implementation under /uncore/agent, but now it is gone. Also, it seems that the code to instantiate L2 cache is missing (I tried to trace versions to several months back, but didn't find a place in the code base to instantiate the L2HellaCacheBank). Is it because I have some misunderstanding of the code? If I want to enable L2 cache in the latest repo, what should I do?

Thanks.

terpstra, 2017-07-17:
Unfortunately, we never had a production-ready L2. Also, the previous L2 spoke a now obsolete version of TileLink. At present there is no L2 in the repository. :-(

closed 2017-07-17 by terpstra

kshalle, 2018-07-07:
Thanks for the note, Wesley.  Intensivate is actually working on porting the L2 to the March 2018 version of Rocket Chip.  We have a high performance, low power, RISC-V core for Enterprise IT.  The core delivers 1.86 IPC at 2.4GHz in 28nm right now, and we're targeting 300mW.  We're filling out the chip, and need an L2.  Porting the previous L2 seems the least effort approach.  

Could you say a bit more about what the challenges were with porting L2 to Chisel 3 and TileLink2?  Any pointers to help us out would be much appreciated.  We are, of course, doing the work open source, and will make the ported L2 available to upstream back into Rocket Chip.

Thanks Wesley,

Sean

terpstra, 2018-07-07:
The L2 we are using at SiFive is a complete rewrite. The fact that we, the people who wrote that old L2, did not try to port it to TL for use in our own chips should tell you something. ;)

We do plan to open-source a simplified variant of the current SiFive L2. The only reason this has not happened yet is because it's not completely trivial and no one has been able to take the time to do this. I believe it is probably also possible to obtain the full version of the L2 under a contract from SiFive.

lennartjanis, 2019-03-06:
Any news on L2 cache support? I would love to have a Rocket-chip with L2 caches that can be taped out.

kshalle, 2019-03-07:
We licensed L2 from SiFive. (It is directory based and can also be used as
coherent LLC, with some additional code wrangling).

LoveZJT, 2019-06-28:
Any news on L2 cache support?

terpstra, 2019-06-28:
Here's the open-source version:
https://github.com/sifive/block-inclusivecache-sifive

waleed-lm, 2020-07-15:
Hey, is there a documents telling how to connect this Inclusive cache with rocket

OrkunAliOzkan, 2023-07-20:
Hi @waleedbinehsan-personal,

Did you have any success finding documentation which explains how to connect the Inclusive cache with the rocket core?

LoveZJT, 2023-07-21:
Here is the doc:
https://chipyard.readthedocs.io/en/stable/Customization/Memory-Hierarchy.html#the-sifive-l2-cache