chipsalliance/rocket-chip #656: Grammatical mistakes infirrtl?

xinjic, 2017-04-06:
While generating verilog in vsim, there are several mistakes about firrtl:
[error] /home/jinxin/Projects/riscv/rocket-chip/firrtl/src/main/scala/firrtl/Emitter.scala:6: object file is not a member of package java.nio
[error] import java.nio.file.{Paths, Files}
[error]                 ^
[error] /home/jinxin/Projects/riscv/rocket-chip/firrtl/src/main/scala/firrtl/StringLit.scala:7: object StandardCharsets is not a member of package java.nio.charset
[error] import java.nio.charset.StandardCharsets.UTF_8
[error]                         ^
[error] /home/jinxin/Projects/riscv/rocket-chip/firrtl/src/main/scala/firrtl/StringLit.scala:25: not found: value UTF_8
[error]             if (c >= 0x20 && c <= 0x7e) new String(Array(c), UTF_8)
[error]                                                              ^
[error] /home/jinxin/Projects/riscv/rocket-chip/firrtl/src/main/scala/firrtl/StringLit.scala:69: object StandardCharsets is not a member of package java.nio.charset
[error]     val byteArray = raw.getBytes(java.nio.charset.StandardCharsets.UTF_8)
[error]                                                   ^
[error] /home/jinxin/Projects/riscv/rocket-chip/firrtl/src/main/scala/firrtl/util/BackendCompilationUtilities.scala:6: object file is not a member of package java.nio
[error] import java.nio.file.Files
[error]                 ^
[error] /home/jinxin/Projects/riscv/rocket-chip/firrtl/src/main/scala/firrtl/util/BackendCompilationUtilities.scala:42: not found: value Files
[error]     Files.createTempDirectory(outer.toPath, timeStamp).toFile
[error]     ^
[error] /home/jinxin/Projects/riscv/rocket-chip/firrtl/src/main/scala/firrtl/util/BackendCompilationUtilities.scala:42: value toPath is not a member of java.io.File
[error]     Files.createTempDirectory(outer.toPath, timeStamp).toFile
[error]                                     ^
[error] 7 errors found
[error] (compile:compileIncremental) Compilation failed
[error] Total time: 29 s, completed Apr 6, 2017 1:51:54 PM
 So,how can I fix it? thks!

terpstra, 2017-04-06:
Could this maybe be a java/scala version issue?
It compiles for me on java 8 and scala 2.11.7.

ucbjrl, 2017-04-06:
It should also work with Java 7.

terpstra, 2017-04-14:
Sounds like you've sorted this out.

closed 2017-04-14 by terpstra