chipsalliance/rocket-chip #657: In VSIM : vcs: command not found make: *** [simv-rocketchip-DefaultConfig] Error 127

ninode, 2017-04-06:
Hello, 

When I run make in the vsim directory, it yields the following:

`mkdir -p /home/alpha/rocket-chip/vsim/generated-src/
cd /home/alpha/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/alpha/rocket-chip/sbt-launch.jar "run-main rocketchip.Generator /home/alpha/rocket-chip/vsim/generated-src rocketchip TestHarness rocketchip DefaultConfig"
[info] Loading project definition from /home/alpha/rocket-chip/project
Using addons: 
[info] Set current project to rocketchip (in build file:/home/alpha/rocket-chip/)
[info] Compiling 1 Scala source to /home/alpha/rocket-chip/chisel3/target/scala-2.11/classes...
[info] Packaging /home/alpha/rocket-chip/chisel3/target/scala-2.11/chisel3_2.11-3.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Running rocketchip.Generator /home/alpha/rocket-chip/vsim/generated-src rocketchip TestHarness rocketchip DefaultConfig
[info] [0.011] Elaborating design...
Interrupt map (2 harts 2 interrupts):
  [1, 2] => ExampleRocketTop

Generated Address Map
	zeros_0 a000000 - c000000, RWX [C]
	plic c000000 - 10000000, RW
	bootrom 10000 - 20000, RX
	ExampleRocketTop 60000000 - 80000000, RWX
	ExampleRocketTop 80000000 - 90000000, RWX [C]
	dmInner 0 - 1000, RWX
	clint 2000000 - 2010000, RW

/dts-v1/;

/ {
	#address-cells = <1>;
	#size-cells = <1>;
	compatible = "ucbbar,rocketchip-unknown-dev";
	model = "ucbbar,rocketchip-unknown";
	L7: cpus {
		#address-cells = <1>;
		#size-cells = <0>;
		timebase-frequency = <1000000>;
		L11: cpu@0 {
			clock-frequency = <0>;
			compatible = "riscv";
			d-cache-block-size = <64>;
			d-cache-sets = <64>;
			d-cache-size = <16384>;
			d-tlb-sets = <1>;
			d-tlb-size = <32>;
			device_type = "cpu";
			i-cache-block-size = <64>;
			i-cache-sets = <64>;
			i-cache-size = <16384>;
			i-tlb-sets = <1>;
			i-tlb-size = <32>;
			mmu-type = "riscv,sv39";
			next-level-cache = <&L4 &L1>;
			reg = <0>;
			riscv,isa = "rv64imafdcs";
			status = "okay";
			tlb-split;
			L12: interrupt-controller {
				#interrupt-cells = <1>;
				compatible = "riscv,cpu-intc";
				interrupt-controller;
			};
		};
	};
	L1: memory@80000000 {
		device_type = "memory";
		reg = <0x80000000 0x10000000>;
	};
	L6: soc {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "ucbbar,rocketchip-unknown-soc", "simple-bus";
		ranges;
		L10: clint@2000000 {
			compatible = "riscv,clint0";
			interrupts-extended = <&L12 3 &L12 7>;
			reg = <0x2000000 0x10000>;
		};
		L8: debug-controller@0 {
			compatible = "sifive,debug-013", "riscv,debug-013";
			interrupts-extended = <&L12 65535>;
			reg = <0x0 0x1000>;
		};
		L9: interrupt-controller@c000000 {
			#address-cells = <0>;
			#interrupt-cells = <1>;
			compatible = "riscv,plic0";
			interrupt-controller;
			interrupts-extended = <&L12 11 &L12 9>;
			reg = <0xc000000 0x4000000>;
			riscv,ndev = <2>;
		};
		L2: mmio@60000000 {
			reg = <0x60000000 0x20000000>;
		};
		L0: offchip-interrupts {
			interrupt-parent = <&L9>;
			interrupts = <1 2>;
		};
		L3: rom@10000 {
			reg = <0x10000 0x10000>;
		};
		L4: rom@a000000 {
			compatible = "ucbbar,cacheable-zero";
			reg = <0xa000000 0x2000000>;
		};
	};
};

[info] [20.367] Done elaborating.
[success] Total time: 46 s, completed Apr 6, 2017 8:28:39 AM
mkdir -p /home/alpha/rocket-chip/vsim/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/alpha/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.fir -o /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.conf
cd /home/alpha/rocket-chip/vsim/generated-src && \
	rm -f /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v && \
	/home/alpha/rocket-chip/vsim/vlsi_mem_gen /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.conf >> /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v.tmp && \
	mv /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v.tmp /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v
cd . && \
	rm -rf csrc && \
	vcs -full64 -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -timescale=1ns/10ps -quiet +rad +v2k +vcs+lic+wait +vc+list -CC "-I/include" -CC "-I/home/alpha/rocket-chip/include" -CC "-std=c++11" -CC "-Wl,-rpath,/home/alpha/rocket-chip/lib" /home/alpha/rocket-chip/lib/libfesvr.so -sverilog +incdir+/home/alpha/rocket-chip/vsim/generated-src +define+CLOCK_PERIOD=1.0 /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.v /home/alpha/rocket-chip/vsim/generated-src/rocketchip.DefaultConfig.behav_srams.v /home/alpha/rocket-chip/vsrc/TestDriver.v /home/alpha/rocket-chip/vsrc/SimDTM.v /home/alpha/rocket-chip/vsrc/jtag_vpi.v /home/alpha/rocket-chip/vsrc/ClockDivider2.v /home/alpha/rocket-chip/vsrc/AsyncResetReg.v  /home/alpha/rocket-chip/csrc/SimDTM.cc /home/alpha/rocket-chip/csrc/jtag_vpi.c +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN +libext+.v  +vpi -P /home/alpha/rocket-chip/vsrc/jtag_vpi.tab -CC "-DVCS_VPI" -o ./simv-rocketchip-DefaultConfig \
	-debug_pp \

/bin/bash: line 2: vcs: command not found
make: *** [simv-rocketchip-DefaultConfig] Error 127
`
Any idea to fix this ? 
Thank you :)

ccelio, 2017-04-06:
> vcs: command not found
> Any idea how to fix this ?

Pay six figures to Synopsys for a license to VCS? :'(

On the plus side, Verilator is provided in rocket-chip/emulator for the free Verilog simulation option.

ninode, 2017-04-06:
Thank you for your response. 
Sorry about that. I'll stick to Verilator then :)

ccelio, 2017-04-06:
Yah... sad story. Unless you have academic connections, Verilator is probably the best option. Good luck!

closed 2017-04-06 by ccelio