chipsalliance/rocket-chip #659: mistakes about java,when generate verilog in vsim

lsanmu, 2017-04-06:
When I try to generate verilog in vsim, there are several mistakes about java. 
[error] (Thread-6) java.io.IOException: Cannot run program "dtc": error=2, No such file or directory
java.io.IOException: Cannot run program "dtc": error=2, No such file or directory
[error] (Thread-1) java.lang.RuntimeException: No exit code: process destroyed.
java.lang.RuntimeException: No exit code: process destroyed.
[error] (run-main-0) java.lang.reflect.InvocationTargetException

And the jdk in my ubuntu14.03 is 
java version "1.8.0_121"
Java(TM) SE Runtime Environment (build 1.8.0_121-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.121-b13, mixed mode)

mwachs5, 2017-04-06:
You need to add Device Tree Compiler package.

lsanmu, 2017-04-07:
Thank you very much. after 
sudo apt-get install Device Tree Compiler
 , the verilog code generated.
Thanks

mwachs5, 2017-04-12:
Closed by #669

closed 2017-04-12 by mwachs5