chipsalliance/rocket-chip #660: Can the c++ emulator run SEPCCPU2000 or SPECCPU2006 benchmark?

Peach92, 2017-04-07:
I see that the c++ emulator can run some benchmarks such as qsort, towers, dhrystone.
I get confused about if it can run some benchmarks such as SPECCPU2000 or SPECCPU2006.
I will very grateful if you can give me some ideas!
Thank you very much. And thanks for your help!

sorear, 2017-04-08:
The C++ emulator operates at an effective clock speed of ~10 kHz and executes no more than ~ 10,000 instructions per second.  While it would be theoretically possible to run SPEC CPU2006, it would take an extremely long time (~3 years for 1T instructions).  Large benchmark runs are better done using one of the FPGA environments.

Peach92, 2017-04-08:
Thank you very much! Thanks for your help!
   I see the README in benchmark, it says, " You cannot use standard clib functions (like memcopy or strcat). You cannot use system calls and thus cannot use printf" and ""You cannot access the simulated command line - ie you cannot use argc and argv within main" and "If you have a very large program and have lots of very big arrays declared on the stack your stack could overwrite your program. Be aware"
   So I still want to know, if I let the emulator run 3 years, can the toolchain compile SPEC CPU2006 source files and can the emulator run the executable program? If yes, please give me some ideas about how to compile SPEC CPU2006.
   Thank you very much! This is very important for my master research paper! I need some research result to decide my master research.

sequencer, 2022-04-08:
Since SPEC is closed source, it cannot be supported from the open-source world.
However we need to consider to provide related tests from some CI machines which has correspond tools.
And test framework is also needed for this issue.

closed 2022-04-08 by sequencer