chipsalliance/rocket-chip #661: Fsim directory and other directories in the slides

ninode, 2017-04-07:
Hello, 
This might be a silly question. And for that I apologize.
In Colin Schmidt's Rocket-chip tutorial. There are directories that I don't find in the rocket-chip git such as fsim, fpga-zynq, Ucore etc...) I'm wondering if there have been new modifcations regarding the folders or should I just install the missing ones. 

Thank you

colinschmidt, 2017-04-07:
There have been a lot of modifications since then. The fpga stuff has been moved out of rocket-chip and lives in ucb-bar/fpga-zynq most of the other folders moved from being git submodules to folders in src/main/scala/

ninode, 2017-04-07:
Thank you for your response. 
So would the old tutorial still apply if I build things accordingly ? Have you made any recent tutorial slides ?

Thank you

closed 2017-09-17 by ninode