chipsalliance/rocket-chip #664: S vs U in ISA string?

sorear, 2017-04-09:
I built rocket-chip and got this in the dts file:

```
riscv,isa = "rv64imafdcs";
```

1. User spec §11.6 says that S is a prefix character which is followed by a one-letter code identifying an extension that is only applicable in S-mode.  By implication a bare "s" at the end of the string does not have a defined meaning.

2. Privileged spec `misa` uses `S` for S-mode support and `U` for U-mode support.  If the ISA string is intended to follow that, then would it also contain a `u`?

terpstra, 2017-04-14:
@aswaterman What to do? I recall at some point you said to not include an 's' at all. Is that still the plan? Note we already have "mmu-type = sv39" so it is redundant.

terpstra, 2017-04-14:
Spike says riscv,isa = "rv64imafdc", so it looks like maybe we should just make rocket-chip do the same thing?

aswaterman, 2017-04-14:
S-mode implies U-mode, so putting 's' in the string but not 'u' is OK, I think.  Maybe the string should be ```rv64imafdcs_sv39```?  I'll also see what @kasanovic thinks.

Regardless, Spike and Rocket should agree.

terpstra, 2017-04-14:
I'd leave the sv39 part off and put it in the (standard) mmu-type field where it already is.

aswaterman, 2017-04-14:
I wrote that because of what the user ISA spec says in Sec. 11.6: "Standard supervisor instruction subsets are defined in Volume II, but are named using “S” as a prefix, followed by a supervisor subset name beginning with a letter and an optional version number."

I can't remember whether we intended Sv39 to fit that definition.

terpstra, 2017-05-17:
This was fixed in #740

closed 2017-05-17 by terpstra