chipsalliance/rocket-chip #667: Hello World problems.

ViccoYang, 2017-04-11:
As a freshman, I follow the steps and make run in emulator.
It make finished without error and I get some .out files, but how can I make sure
the result is right?

Another problem, if I write a simple custom test hello.c
`#include<stdio.h>
int main(){
    printf("Hello RISCV\n");
    return 0;
}`
Is there any tutorial about compling and running custom .c files in rocket chip?

Thanks for you help!
Vicco 04112017

ccelio, 2017-04-11:
For `make run`, each test will print a failure string if it failed (and provide bash a failure code). If everything worked, it will finish up quietly.

For compiling and running your Hello World on spike ISA simulator:
https://github.com/riscv/riscv-tools#-testing-your-toolchain

Then replace `spike` with your rocket-chip emulator: `./rocketchip-AwesomeConfig pk hello`

You can also look more closely at the `benchmarks` part of RISC-V tests, which builds C programs that run bare-metal (https://github.com/riscv/riscv-tests/tree/6a1a38d421fd3e24bdc179d58d33572636b903b2/benchmarks). Look at the Makefile, and the `common` directory to see how the sausage is made if you want to avoid using the pk.

closed 2017-04-11 by ccelio

ViccoYang, 2017-04-13:
Thanks for your help and I can run my own testcase now, but how can I know how many cycles the testcase takes?