chipsalliance/rocket-chip #668: Java Errors while generating Verilog code for FPGA

kkiiopq, 2017-04-11:
Hi,

I was trying to generate Verilog code for FPGA, and followed all the steps in the introduction. Anyway I 
met this:

cy@cy-Lenovo-IdeaPad-Y410P:~/rocket-chip/vsim$ make verilog CONFIG=TinyConfig
make -C /home/cy/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -XX:MaxHeapSize=512m -jar /home/cy/rocket-chip/sbt-launch.jar" root_dir=/home/cy/rocket-chip/firrtl build-scala
make[1]: Entering directory `/home/cy/rocket-chip/firrtl'
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -XX:MaxHeapSize=512m -jar /home/cy/rocket-chip/sbt-launch.jar "assembly"
[info] Loading project definition from /home/cy/rocket-chip/firrtl/project
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
[info] Set current project to firrtl (in build file:/home/cy/rocket-chip/firrtl/)
[info] Compiling 73 Scala sources and 4 Java sources to /home/cy/rocket-chip/firrtl/target/scala-2.11/classes...
[warn] /home/cy/rocket-chip/firrtl/src/main/scala/firrtl/Utils.scala:403: unreachable code
[warn]       case (AnalogType(w1), AnalogType(w2)) => Nil
[warn]                                                ^
[warn] there were 16 deprecation warnings; re-run with -deprecation for details
[warn] there were 22 feature warnings; re-run with -feature for details
[warn] three warnings found
java.io.IOException: Cannot run program "javac" (in directory "/home/cy/rocket-chip/firrtl"): error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1047)
	at sbt.SimpleProcessBuilder.run(ProcessImpl.scala:349)
	at sbt.AbstractProcessBuilder.run(ProcessImpl.scala:128)
	at sbt.AbstractProcessBuilder$$anonfun$runBuffered$1.apply(ProcessImpl.scala:159)
	at sbt.AbstractProcessBuilder$$anonfun$runBuffered$1.apply(ProcessImpl.scala:159)
	at sbt.compiler.javac.JavacLogger.buffer(JavacProcessLogger.scala:31)
	at sbt.AbstractProcessBuilder.runBuffered(ProcessImpl.scala:159)
	at sbt.AbstractProcessBuilder.$bang(ProcessImpl.scala:156)
	at sbt.compiler.javac.ForkedJava$$anonfun$launch$1.apply(ForkedJava.scala:24)
	at sbt.compiler.javac.ForkedJava$$anonfun$launch$1.apply(ForkedJava.scala:17)
	at sbt.compiler.javac.ForkedJava$$anonfun$withArgumentFile$1.apply(ForkedJava.scala:47)
	at sbt.compiler.javac.ForkedJava$$anonfun$withArgumentFile$1.apply(ForkedJava.scala:44)
	at sbt.IO$.withTemporaryDirectory(IO.scala:344)
	at sbt.compiler.javac.ForkedJava$.withArgumentFile(ForkedJava.scala:44)
	at sbt.compiler.javac.ForkedJava$.launch(ForkedJava.scala:17)
	at sbt.compiler.javac.ForkedJavaCompiler.run(ForkedJava.scala:68)
	at sbt.compiler.javac.JavaCompilerAdapter.compileWithReporter(JavaCompilerAdapter.scala:31)
	at sbt.compiler.javac.AnalyzingJavaCompiler$$anonfun$compile$1.apply$mcV$sp(AnalyzingJavaCompiler.scala:65)
	at sbt.compiler.javac.AnalyzingJavaCompiler$$anonfun$compile$1.apply(AnalyzingJavaCompiler.scala:65)
	at sbt.compiler.javac.AnalyzingJavaCompiler$$anonfun$compile$1.apply(AnalyzingJavaCompiler.scala:65)
	at sbt.compiler.javac.AnalyzingJavaCompiler.timed(AnalyzingJavaCompiler.scala:93)
	at sbt.compiler.javac.AnalyzingJavaCompiler.compile(AnalyzingJavaCompiler.scala:64)
	at sbt.compiler.MixedAnalyzingCompiler$$anonfun$compileJava$1$1.apply$mcV$sp(MixedAnalyzingCompiler.scala:60)
	at sbt.compiler.MixedAnalyzingCompiler$$anonfun$compileJava$1$1.apply(MixedAnalyzingCompiler.scala:60)
	at sbt.compiler.MixedAnalyzingCompiler$$anonfun$compileJava$1$1.apply(MixedAnalyzingCompiler.scala:60)
	at sbt.compiler.MixedAnalyzingCompiler.timed(MixedAnalyzingCompiler.scala:74)
	at sbt.compiler.MixedAnalyzingCompiler.compileJava$1(MixedAnalyzingCompiler.scala:59)
	at sbt.compiler.MixedAnalyzingCompiler.compile(MixedAnalyzingCompiler.scala:64)
	at sbt.compiler.IC$$anonfun$compileInternal$1.apply(IncrementalCompiler.scala:160)
	at sbt.compiler.IC$$anonfun$compileInternal$1.apply(IncrementalCompiler.scala:160)
	at sbt.inc.IncrementalCompile$$anonfun$doCompile$1.apply(Compile.scala:66)
	at sbt.inc.IncrementalCompile$$anonfun$doCompile$1.apply(Compile.scala:64)
	at sbt.inc.IncrementalCommon.cycle(IncrementalCommon.scala:32)
	at sbt.inc.Incremental$$anonfun$1.apply(Incremental.scala:72)
	at sbt.inc.Incremental$$anonfun$1.apply(Incremental.scala:71)
	at sbt.inc.Incremental$.manageClassfiles(Incremental.scala:99)
	at sbt.inc.Incremental$.compile(Incremental.scala:71)
	at sbt.inc.IncrementalCompile$.apply(Compile.scala:54)
	at sbt.compiler.IC$.compileInternal(IncrementalCompiler.scala:160)
	at sbt.compiler.IC$.incrementalCompile(IncrementalCompiler.scala:138)
	at sbt.Compiler$.compile(Compiler.scala:152)
	at sbt.Compiler$.compile(Compiler.scala:138)
	at sbt.Defaults$.sbt$Defaults$$compileIncrementalTaskImpl(Defaults.scala:860)
	at sbt.Defaults$$anonfun$compileIncrementalTask$1.apply(Defaults.scala:851)
	at sbt.Defaults$$anonfun$compileIncrementalTask$1.apply(Defaults.scala:849)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
[... 65 lines cut ...]
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
	at sbt.Execute.work(Execute.scala:237)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1.apply(Execute.scala:228)
	at sbt.ConcurrentRestrictions$$anon$4$$anonfun$1.apply(ConcurrentRestrictions.scala:159)
	at sbt.CompletionService$$anon$2.call(CompletionService.scala:28)
	at java.util.concurrent.FutureTask.run(FutureTask.java:262)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:471)
	at java.util.concurrent.FutureTask.run(FutureTask.java:262)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:615)
	at java.lang.Thread.run(Thread.java:745)
[error] (compile:compileIncremental) java.io.IOException: Cannot run program "javac" (in directory "/home/cy/rocket-chip/firrtl"): error=2, No such file or directory
[error] Total time: 38 s, completed Apr 11, 2017 7:40:12 PM
make[1]: *** [/home/cy/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 1
make[1]: Leaving directory /home/cy/rocket-chip/firrtl
make: *** [/home/cy/rocket-chip/firrtl/utils/bin/firrtl.jar] Error 2

Hope u can help there, thanks a lot!

Best regards,
Cy

mwachs5, 2017-04-11:
You need to install Java SDK to get 'javac' command

mwachs5, 2017-04-12:
Closed by #669

closed 2017-04-12 by mwachs5

kkiiopq, 2017-04-12:
I think I have JAVA OpenJDK installed on my computer by checking this:
cy@cy-Lenovo-IdeaPad-Y410P:~$ java -version
java version "1.7.0_121"
OpenJDK Runtime Environment (IcedTea 2.6.8) (7u121-2.6.8-1ubuntu0.14.04.3)
OpenJDK Server VM (build 24.121-b00, mixed mode)