chipsalliance/rocket-chip #669: Package Dependency List(s) should be added to README

mwachs5, 2017-04-11:
There isn't a package dependency list in this repo, though users need to install various things like javac and dtc. Some may be covered by Chisel3 / Firrtl README, but the latter probably isn't. Should we add something to the README about this?

aswaterman, 2017-04-11:
riscv-tools' README has the package list depended upon by the toolchain (including dtc).  To avoid having to keep them in sync, maybe we have rocket-chip's README refer to that and Chisel's READMEs?

mwachs5, 2017-04-11:
Yes, I think we should refer to the other READMEs if they exist. But it seems like DTC is our own dependency?

mwachs5, 2017-04-11:
Oh sorry I retract my statement if riscv-tools includes dtc

aswaterman, 2017-04-11:
yeah i just checked, it does

mwachs5, 2017-04-12:
closed by #670

closed 2017-04-12 by mwachs5

sorear, 2017-04-20:
rocket-chip master depends on dtc.

riscv-tools priv-1.10 README mentions dtc.

riscv-tools master README does NOT mention dtc.

rocket-chip master has a README which points at riscv-tools' master README.

_If you follow links from the rocket-chip master README, you will not install all of the tools needed to build rocket-chip master._