chipsalliance/rocket-chip #674: [emulator-rocketchip-DefaultConfig-debug] error

ninode, 2017-04-12:
Hello, 

I'm following the instruction to using the verilator.

Make run-debug yields the following errors. Any idea what I'm missing ?

Thank you !

```
mkdir -p /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug/rocketchip.DefaultConfig
/home/alpha/fpga-zynq/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/alpha/fpga-zynq/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/alpha/fpga-zynq/rocket-chip/riscv-tools/include -I/home/alpha/fpga-zynq/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/alpha/fpga-zynq/rocket-chip/csrc/verilator.h" -Mdir /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug/rocketchip.DefaultConfig  --trace \
	-o /home/alpha/fpga-zynq/rocket-chip/emulator/emulator-rocketchip-DefaultConfig-debug /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug/rocketchip.DefaultConfig.v /home/alpha/fpga-zynq/rocket-chip/csrc/emulator.cc /home/alpha/fpga-zynq/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/alpha/fpga-zynq/rocket-chip/riscv-tools/lib -Wl,-rpath,/home/alpha/fpga-zynq/rocket-chip/riscv-tools/lib -L/home/alpha/fpga-zynq/rocket-chip/emulator -lfesvr -lpthread" \
	-CFLAGS "-I/home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug -include /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug/rocketchip.DefaultConfig/VTestHarness.h"
make VM_PARALLEL_BUILDS=1 -C /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug/rocketchip.DefaultConfig -f VTestHarness.mk
make[1]: Entering directory `/home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug/rocketchip.DefaultConfig'
VTestHarness.mk:63: /home/alpha/rocket-chip/emulator/verilator/install/share/verilator/include/verilated.mk: No such file or directory
make[1]: *** No rule to make target `/home/alpha/rocket-chip/emulator/verilator/install/share/verilator/include/verilated.mk'.  Stop.
make[1]: Leaving directory `/home/alpha/fpga-zynq/rocket-chip/emulator/generated-src-debug/rocketchip.DefaultConfig'
make: *** [emulator-rocketchip-DefaultConfig-debug] Error 2

```

mwachs5, 2017-04-12:
Seems that some environment variable is set which shouldn't be... it should be installing verilator into the current emulator directory and using that `/home/alpha/fpga-zynq/rocket-chip`, but it looks like the generated makefile VTestHarness.mk is looking in `/home/alpha/rocket-chip.`

Try unsetting any verilator related path variables, you should just need RISCV variable set to your tools install location. Make clean and try make run again.

Note that Verilator is installed automatically by the 'make run' target, you shouldn't need to manually install it.

ninode, 2017-05-04:
Thank you for replying. 

I have run make clean and set the RISCV environment variable only. No other verilator parameters are set. 

I still run into an error after the command: make in emulator

```
[success] Total time: 53 s, completed May 4, 2017 10:04:36 AM
mkdir -p /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/alpha/fpga-zynq/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.fir -o /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v -X verilog
mkdir -p /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig
/home/alpha/fpga-zynq/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/alpha/fpga-zynq/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/fpga-zynq/rocket-chip/riscv-tools/include -I/home/alpha/fpga-zynq/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/alpha/fpga-zynq/rocket-chip/csrc/verilator.h" -Mdir /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig \
	-o /home/alpha/fpga-zynq/rocket-chip/emulator/emulator-rocketchip-DefaultConfig /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /home/alpha/fpga-zynq/rocket-chip/csrc/emulator.cc /home/alpha/fpga-zynq/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/fpga-zynq/rocket-chip/riscv-tools/lib -Wl,-rpath,/home/fpga-zynq/rocket-chip/riscv-tools/lib -L/home/alpha/fpga-zynq/rocket-chip/emulator -lfesvr -lpthread" \
	-CFLAGS "-I/home/alpha/fpga-zynq/rocket-chip/emulator/generated-src -include /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h"
make VM_PARALLEL_BUILDS=1 -C /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig -f VTestHarness.mk
make[1]: Entering directory `/home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig'
g++  -I.  -MMD -I/home/alpha/fpga-zynq/rocket-chip/emulator/verilator/install/share/verilator/include -I/home/alpha/fpga-zynq/rocket-chip/emulator/verilator/install/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_TRACE=0 -DVM_COVERAGE=0 -Wno-char-subscripts -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable     -I/home/alpha/fpga-zynq/rocket-chip/emulator/generated-src -include /home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h -O1 -std=c++11 -I/home/fpga-zynq/rocket-chip/riscv-tools/include -I/home/alpha/fpga-zynq/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/alpha/fpga-zynq/rocket-chip/csrc/verilator.h   -c -o SimDTM.o /home/alpha/fpga-zynq/rocket-chip/csrc/SimDTM.cc
/home/alpha/fpga-zynq/rocket-chip/csrc/SimDTM.cc:3:23: fatal error: fesvr/dtm.h: No such file or directory
 #include <fesvr/dtm.h>
                       ^
compilation terminated.
make[1]: *** [SimDTM.o] Error 1
make[1]: Leaving directory `/home/alpha/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig'
make: *** [emulator-rocketchip-DefaultConfig] Error 2
```
and then make run in emulator gives the following:
```

alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/emulator$ make -j8 run
make: *** No rule to make target `/home/fpga-zynq/rocket-chip/riscv-tools/riscv64-unknown-elf/share/riscv-tests/isa/rv64um-v-mul', needed by `output/rv64um-v-mul'.  Stop.
```

The rocket-chip repo is up-to-date (git pull origin master). Should I also update the riscv-tools repo from master also ? 

Thank you

terpstra, 2017-05-17:
You have not built a full toolchain. Run ./build.sh in rocket-chip/riscv-tools with your desired RISCV set.

closed 2017-09-17 by ninode