chipsalliance/rocket-chip #681: build of emulator under windows 10 linux subsystem fails.

keskella, 2017-04-14:
On windows 10 Linux subsystem, build of emulator fails with the following
Steps taken before error: followed through verilator install from git, as per chisel page.  Built riscv-tools with more luck then anything else (beats me, it was broken for a month or so, and then worked.  I'm guilty of trying all the knobs).

:
parabellum@DESKTOP-410PS4S:/mnt/c/Users/jchay/Documents/GitHub/rocket-chip/emulator$ make
mkdir -p /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/emulator/generated-src/
cd /mnt/c/Users/jchay/Documents/GitHub/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /mnt/c/Users/jchay/Do
cuments/GitHub/rocket-chip/sbt-launch.jar "run-main rocketchip.Generator /mnt/c/Users/jchay/Documents/GitHub/rocket-chip
/emulator/generated-src rocketchip TestHarness rocketchip DefaultConfig"
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/project
Using addons:
[info] Set current project to rocketchip (in build file:/mnt/c/Users/jchay/Documents/GitHub/rocket-chip/)
[info] Compiling 1 Scala source to /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/chisel3/target/scala-2.11/classes...
[info] Packaging /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/chisel3/target/scala-2.11/chisel3_2.11-3.1-SNAPSHOT.jar
 ...
[info] Done packaging.
[info] Compiling 217 Scala sources to /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/target/scala-2.11/classes...
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/groundtest/DummyPTW.scala:33: value asTypeOf is n
ot a member of chisel3.core.UInt
[error]   val s2_resp = Wire(init = 0.U.asTypeOf(new PTWResp))
[error]                                 ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/groundtest/DummyPTW.scala:49: value asTypeOf is n
ot a member of chisel3.core.UInt
[error]     requestor.status := 0.U.asTypeOf(requestor.status)
[error]                             ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/CSR.scala:381: value asTypeOf is not a mem
ber of chisel3.core.UInt
[error]     val read_sstatus = Wire(init = 0.U.asTypeOf(new MStatus))
[error]                                        ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/CSR.scala:425: value asTypeOf is not a mem
ber of chisel3.core.UInt
[error]     val read_pmp = reg_pmp.padTo(CSR.maxPMPs, 0.U.asTypeOf(new PMP))
[error]                                                   ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/CSR.scala:611: value asTypeOf is not a mem
ber of chisel3.core.UInt
[error]       val new_mip = readModifyWriteCSR(io.rw.cmd, reg_mip.asUInt, io.rw.wdata).asTypeOf(new MIP)
[error]                                                                                ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/PMP.scala:123: value asTypeOf is not a mem
ber of chisel3.core.UInt
[error]     ((true.B, 0.U.asTypeOf(new PMP)) /: pmps) { case ((h, prev), pmp) =>
[error]                   ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/PMP.scala:142: value asTypeOf is not a mem
ber of chisel3.core.UInt
[error]   val pmp0 = Wire(init = 0.U.asTypeOf(new PMP))
[error]                              ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/Rocket.scala:7: object withReset is not a
member of package chisel3.core
[error] import chisel3.core.withReset
[error]        ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/Rocket.scala:176: not found: value withRes
et
[error]   ibuf.io.imem <> (if (usingCompressed) withReset(reset || take_pc) { Queue(io.imem.resp, 1, flow = true) } else
 io.imem.resp)
[error]                                         ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:79: value asTypeOf is not a memb
er of Chisel.UInt
[error]   val entries = reg_entries.map(_.asTypeOf(new Entry))
[error]                                   ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:98: value ppn is not a member of
 Nothing
[error]                 Mux(vm_enabled, entries.last.ppn, vpn(ppnBits-1, 0)))
[error]                                              ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:119: value level is not a member
 of Nothing
[error]       tagMatch = tagMatch && (entries(i).level < j || entries(i).tag(base + pgLevelBits - 1, base) === vpn(base
+ pgLevelBits - 1, base))
[error]                                          ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:119: value tag is not a member o
f Nothing
[error]       tagMatch = tagMatch && (entries(i).level < j || entries(i).tag(base + pgLevelBits - 1, base) === vpn(base
+ pgLevelBits - 1, base))
[error]                                                                  ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:124: value level is not a member
 of Nothing
[error]   val level = Mux1H(hitsVec.init, entries.map(_.level))
[error]                                                 ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:125: value ppn is not a member o
f Nothing
[error]   val partialPPN = Mux1H(hitsVec.init, entries.map(_.ppn))
[error]                                                      ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:162: value ae is not a member of
 Nothing
[error]   val priv_ok = entries.map(_.ae).asUInt | Mux(priv_s, ~Mux(io.ptw.status.sum, UInt(0), entries.map(_.u).asUInt)
, entries.map(_.u).asUInt)
[error]                               ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:162: value u is not a member of
Nothing
[error]   val priv_ok = entries.map(_.ae).asUInt | Mux(priv_s, ~Mux(io.ptw.status.sum, UInt(0), entries.map(_.u).asUInt)
, entries.map(_.u).asUInt)
[error]                                                                                                       ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:162: value u is not a member of
Nothing
[error]   val priv_ok = entries.map(_.ae).asUInt | Mux(priv_s, ~Mux(io.ptw.status.sum, UInt(0), entries.map(_.u).asUInt)
, entries.map(_.u).asUInt)
[error]
                ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:166: value pr is not a member of
 Nothing
[error]   val pr_array = Cat(Fill(2, prot_r), entries.init.map(_.pr).asUInt)
[error]                                                          ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:167: value pw is not a member of
 Nothing
[error]   val pw_array = Cat(Fill(2, prot_w), entries.init.map(_.pw).asUInt)
[error]                                                          ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:168: value px is not a member of
 Nothing
[error]   val px_array = Cat(Fill(2, prot_x), entries.init.map(_.px).asUInt)
[error]                                                          ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:169: value c is not a member of
Nothing
[error]   val c_array = Cat(Fill(2, cacheable), entries.init.map(_.c).asUInt)
[error]                                                            ^
[error] /mnt/c/Users/jchay/Documents/GitHub/rocket-chip/src/main/scala/rocket/TLB.scala:224: value g is not a member of
Nothing
[error]                Mux(io.req.bits.sfence.bits.rs2, valid & entries.map(_.g).asUInt, 0))
[error]                                                                       ^
[error] 23 errors found
[error] (rocketchip/compile:compileIncremental) Compilation failed
[error] Total time: 141 s, completed Apr 14, 2017 7:25:07 AM
Makefrag-verilator:12: recipe for target '/mnt/c/Users/jchay/Documents/GitHub/rocket-chip/emulator/generated-src/rocketc
hip.DefaultConfig.fir' failed
make: *** [/mnt/c/Users/jchay/Documents/GitHub/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.fir] Error 1

sorear, 2017-04-14:
```
    Triple ` on a line before and after marks verbatim blocks in
    Markdown and GitHub's markdown-based comment language
    ```

terpstra, 2017-04-15:
It sounds like you either do not have your submodules up-to-date (git submodule update --recursive) or you are on a strange branch. I doubt this has anything to do with Windows.

closed 2022-04-08 by sequencer