chipsalliance/rocket-chip #686: Error trying to Make the emulator on Mac

arechap, 2017-04-19:
Hello All,

So I've been trying to get rocket-chip running on my Mac and everything was going well until I try to run the emulator. After the build.sh script completes I go into the emulator folder and type "make" which fails with the following errors:

`In file included from ../V3ParseLex.cpp:37:
V3Lexer_pregen.yy.cpp:3553:8: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
                        yyin.rdbuf(std::cin.rdbuf());
                        ~~~~^
                            ->
V3Lexer_pregen.yy.cpp:3560:9: error: member reference type 'std::ostream *' (aka 'basic_ostream<char> *') is a pointer; did you mean to use '->'?
                        yyout.rdbuf(std::cout.rdbuf());
                        ~~~~~^
                             ->
V3Lexer_pregen.yy.cpp:7056:50: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
                        YY_CURRENT_BUFFER_LVALUE->yy_input_file = yyin.rdbuf();
                                                                  ~~~~^
                                                                      ->
V3Lexer_pregen.yy.cpp:7176:2: error: cannot initialize a member subobject of type 'std::istream *' (aka 'basic_istream<char> *') with an rvalue of type 'basic_streambuf<char_type, traits_type> *'
      (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyin(arg_yyin ? arg_yyin->rdbuf() : std::cin.rdbuf()),
        ^    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
V3Lexer_pregen.yy.cpp:7177:2: error: cannot initialize a member subobject of type 'std::ostream *' (aka 'basic_ostream<char> *') with an rvalue of type 'basic_streambuf<char_type, traits_type> *'
      (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyout(arg_yyout ? arg_yyout->rdbuf() : std::cout.rdbuf())
        ^     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
V3Lexer_pregen.yy.cpp:7179:2: error: use of undeclared identifier 'ctor_common'
        ctor_common();
        ^
V3Lexer_pregen.yy.cpp:7184:14: error: out-of-line definition of 'V3LexerBase' does not match any declaration in 'V3LexerBase'
yyFlexLexer::yyFlexLexer( std::istream& arg_yyin, std::ostream& arg_yyout ):
             ^~~~~~~~~~~
../V3ParseLex.cpp:36:21: note: expanded from macro 'yyFlexLexer'
#define yyFlexLexer V3LexerBase
                    ^~~~~~~~~~~
V3Lexer_pregen.yy.cpp:7175:27: note: type of 1st parameter of member declaration does not match definition ('std::istream *' (aka 'basic_istream<char> *') vs 'std::istream &' (aka 'basic_istream<char> &'))
yyFlexLexer::yyFlexLexer( FLEX_STD istream* arg_yyin, FLEX_STD ostream* arg_yyout ):
                          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/FlexLexer.h:53:22: note: expanded from macro 'FLEX_STD'
#    define FLEX_STD std::
                     ^
In file included from ../V3ParseLex.cpp:37:
V3Lexer_pregen.yy.cpp:7185:2: error: cannot initialize a member subobject of type 'std::istream *' (aka 'basic_istream<char> *') with an rvalue of type 'basic_streambuf<char_type, traits_type> *'
      (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyin(arg_yyin.rdbuf()),
        ^    ~~~~~~~~~~~~~~~~
V3Lexer_pregen.yy.cpp:7186:2: error: cannot initialize a member subobject of type 'std::ostream *' (aka 'basic_ostream<char> *') with an rvalue of type 'basic_streambuf<char_type, traits_type> *'
      (aka 'basic_streambuf<char, std::__1::char_traits<char> > *')
        yyout(arg_yyout.rdbuf())
        ^     ~~~~~~~~~~~~~~~~~
V3Lexer_pregen.yy.cpp:7188:2: error: use of undeclared identifier 'ctor_common'
        ctor_common();
        ^
V3Lexer_pregen.yy.cpp:7193:19: error: out-of-line definition of 'ctor_common' does not match any declaration in 'V3LexerBase'
void yyFlexLexer::ctor_common()
                  ^~~~~~~~~~~
V3Lexer_pregen.yy.cpp:7231:19: error: out-of-line definition of 'switch_streams' does not match any declaration in 'V3LexerBase'
void yyFlexLexer::switch_streams( std::istream& new_in, std::ostream& new_out )
                  ^~~~~~~~~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/FlexLexer.h:129:31: note: type of 1st parameter of member declaration does not match definition ('std::istream *'
      (aka 'basic_istream<char> *') vs 'std::istream &' (aka 'basic_istream<char> &'))
        virtual void switch_streams( FLEX_STD istream* new_in, FLEX_STD ostream* new_out = 0 );
                                     ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/FlexLexer.h:53:22: note: expanded from macro 'FLEX_STD'
#    define FLEX_STD std::
                     ^
In file included from ../V3ParseLex.cpp:37:
V3Lexer_pregen.yy.cpp:7235:41: error: no viable conversion from 'std::istream' (aka 'basic_istream<char>') to 'std::istream *' (aka 'basic_istream<char> *')
        yy_switch_to_buffer( yy_create_buffer( new_in, YY_BUF_SIZE  ) );
                                               ^~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/ios:585:9: note: candidate function
        operator bool() const {return !fail();}
        ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/FlexLexer.h:121:62: note: passing argument to parameter 's' here
        struct yy_buffer_state* yy_create_buffer( FLEX_STD istream* s, int size );
                                                                    ^
In file included from ../V3ParseLex.cpp:37:
V3Lexer_pregen.yy.cpp:7238:7: error: member reference type 'std::ostream *' (aka 'basic_ostream<char> *') is a pointer; did you mean to use '->'?
        yyout.rdbuf(new_out.rdbuf());
        ~~~~~^
             ->
V3Lexer_pregen.yy.cpp:7246:10: error: assigning to 'std::istream *' (aka 'basic_istream<char> *') from incompatible type 'std::istream **' (aka 'basic_istream<char> **'); remove &
                new_in = &yyin;
                       ^ ~~~~~
V3Lexer_pregen.yy.cpp:7250:11: error: assigning to 'std::ostream *' (aka 'basic_ostream<char> *') from incompatible type 'std::ostream **' (aka 'basic_ostream<char> **'); remove &
                new_out = &yyout;
                        ^ ~~~~~~
V3Lexer_pregen.yy.cpp:7253:17: error: no viable conversion from 'std::istream' (aka 'basic_istream<char>') to 'std::istream *' (aka 'basic_istream<char> *')
        switch_streams(*new_in, *new_out);
                       ^~~~~~~
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/ios:585:9: note: candidate function
        operator bool() const {return !fail();}
        ^
V3Lexer_pregen.yy.cpp:7243:49: note: passing argument to parameter 'new_in' here
void yyFlexLexer::switch_streams( std::istream* new_in, std::ostream* new_out )
                                                ^
V3Lexer_pregen.yy.cpp:7257:18: error: out-of-line definition of 'LexerInput' does not match any declaration in 'V3LexerBase'
int yyFlexLexer::LexerInput( char* buf, int /* max_size */ )
                 ^~~~~~~~~~
V3Lexer_pregen.yy.cpp:7262:11: error: member reference type 'std::istream *' (aka 'basic_istream<char> *') is a pointer; did you mean to use '->'?
        if ( yyin.eof() || yyin.fail() )
             ~~~~^
                 ->
fatal error: too many errors emitted, stopping now [-ferror-limit=]`

My suspicion is that this has to do with Mac using clang by default and I need to switch to gcc/g++. But I just wanted to bring this to someone's attention in case the error goes deeper than that.

Thank you for your time.

sorear, 2017-04-19:
```
    To make log fragments and other preformatted text legible,
    wrap them in triple backquotes as demonstrated here.
    ```

arechap, 2017-04-19:
Thanks for the tip! It looks much nicer now.

arechap, 2017-04-19:
OK quick update! so it looks like running "make" in the emulator folder actually finishes. It still shows all the same errors as above, but it somehow pushes through and ends up printing:
```[info] [9.762] Done elaborating.
[success] Total time: 22 s, completed Apr 18, 2017 11:19:08 PM```

But then when I try 'make -j4 run' or any other test I get the same error message as before and does not push through to finish the way just regular "make" does.

arechap, 2017-04-19:
Another update, everything works nicely on an Ubuntu machine so this seems to be a Mac specific issue.

arechap, 2017-04-20:
Ok I have a fix! Here's what I had to do:
1. used Homebrew to install gcc/g++
2. create a symlink in /usr/local/bin so that every time gcc/g++ is called it points to the homebrew version instead of the Mac clang/llvm version.
3. used Homebrew to install verilator on my own.
4. Modified Makefrag-verilator in the emulator folder in the following way:
`
# Build and install our own Verilator, to work around versionining issues.
#VERILATOR_VERSION=3.884
#VERILATOR_SRCDIR ?= verilator/src/verilator-$(VERILATOR_VERSION)
#VERILATOR_TARGET := $(abspath verilator/install/bin/verilator)
#INSTALLED_VERILATOR ?= $(VERILATOR_TARGET)
#$(VERILATOR_TARGET): $(VERILATOR_SRCDIR)/bin/verilator
#	$(MAKE) -C $(VERILATOR_SRCDIR) installbin installdata
#	touch $@

INSTALLED_VERILATOR = /usr/local/bin/verilator
`

So those are lines 23-32 and basically what I've done is told rocket chip to use the verilator I installed on my own as opposed to the provided verilator. 

And after those steps it works! "make" completes and so does "make -j8 run"! I will leave this open because of that comment "Build and install our own Verilator, to work around versionining issues." I wouldn't be surprised if I run into a problem sometime during this project. 

If I do run into an issue I'll just continue this thread, but hopefully you won't hear from me again! (but easier mac support would be greatly appreciated)

jaggies, 2018-03-31:
This is much easier, assuming you have a path to the gcc binaries in your shell.  You can substitute absolute paths, of course:

```
autoconf
./configure CC=`which gcc-mp-4.9` CXX=`which g++-mp-4.9`
make
make test
sudo make install
```

sequencer, 2022-04-08:
related to #2953 need document on multiple platform supporting.

closed 2022-04-08 by sequencer